bitaxeorg/ESP-Miner #1: getting issue on reading the DS4432U

developeralgo8888, 2023-01-17:
```
I (1947) wifi station: connected to ap SSID:ABHomie
I (1957) i2c-test: Init LEDs!
I (1957) i2c-test: I2C initialized successfully
ESP_ERROR_CHECK failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x42008e62
0x42008e62: DS4432U_read at /developer/Downloads/ESP-Miner-i2c_test/build/../main/DS4432U.c:75 (discriminator 1)

file: "../main/DS4432U.c" line 75
func: DS4432U_read
expression: register_read(DS4432U_SENSOR_ADDR, DS4432U_OUT1_REG, data, 1)

abort() was called at PC 0x4037de33 on core 0
0x4037de33: _esp_error_check_failed at /developer/.espressif/esp-idf-ae062fbba3ded0aa/v5.0/components/esp_system/esp_err.c:47
```

skot, 2023-01-17:
I think that is ESP-IDF's lovely way of saying no response from the I2C address `DS4432U_SENSOR_ADDR`.

What do you have `DS4432U_SENSOR_ADDR` set to? Is is `0x48` ?
Can you confirm the DS4432U+ has power and your I2C lines SDA and SCL are connected properly? You should get a 0x00 response.

developeralgo8888, 2023-01-17:
I think it DS4432U_SENSOR_ADDR set to 0x48. Let me confirm

developeralgo8888, 2023-01-18:
issue solved

closed 2023-01-18 by developeralgo8888