bitaxeorg/ESP-Miner #5: How can I flash it on a esp32

a-1500, 2023-03-20:
How can I flash it on a esp32, I have no idea how, please explain

skot, 2023-03-20:
The ESP32-S3 on the bitaxe is [flashed just like any other ESP32](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html) kit or dev board out there. 

Just be aware you'll need to use a [ESP-Prog](https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-iot-solution/en/latest/hw-reference/ESP-Prog_guide.html) and a [Tag Connect cable](https://www.tag-connect.com/product/tc2030-idc-nl), as there is no USB port on the bitaxe.

a-1500, 2023-03-20:
I meant how can I flash the esp-miner

skot, 2023-03-20:
Sorry, I'm not sure what you are asking.

a-1500, 2023-03-20:
> Sorry, I'm not sure what you are asking.

Can I flash the firmware ESP-miner on the esp32 on the bimax bord to be able to mine

a-1500, 2023-03-20:
> This firmware needs _alot_ more work before it will mine. You can see the latest if you checkout the `i2c_test` branch.

Ok, and I'm very excited for the firmware to be completed

closed 2023-05-25 by skot