bitaxeorg/ESP-Miner #31: Support bm1397 in mode_1

BONIIXS, 2023-09-10:
Hey there I just wanted to ask if it was possible to use the firmware with a custom miner with the same hardware but the BM1397 will be in mode 1, is it possible?

skot, 2023-09-10:
Yes, definitely! AFAIK esp-miner doesn't care what mode the ASIC is in.

johnny9, 2023-09-10:
What does mode 1 refer to?

skot, 2023-09-10:
"Mode" in this context is a pin on the ASIC that reverses the side the data input and output pins are on. It dramatically simplifies PCB routing.

BONIIXS, 2023-09-10:
Thanks for the answer skot

BONIIXS, 2023-09-10:
Thanks for the answer but does it also support chaining(multiple asic chips in series)

skot, 2023-09-11:
we have added theoretical support for multiple ASICs to esp-miner. But until one of the hex designs is done it's untested.

BONIIXS, 2023-09-11:
I think I can make one

BONIIXS, 2023-09-11:
I've just started, and I think it'll take me about 3 to 7 days to finish. It's going to be like a hexagonal-designed battle axe. how many ASIC chips I should use?

closed 2023-09-24 by johnny9