bitaxeorg/ESP-Miner #36: Not booting after compiling

ed0nkey, 2023-09-21:
I've did everything that the guide said and even tried via vscode too, but after flashing using the latest project files it fails to boot. I don't have the slightest idea how this work but using the 1.0 release it boots fine.
[image]

johnny9, 2023-09-21:
It looks like it didn't flash the image correctly. Just the bootloader is there.

ed0nkey, 2023-09-22:
It is very strange because I saw everything being flashed correctly after compiling and I really don't know what could be causing this issue. Other branches like i2c_test flashes and boot just fine.

ed0nkey, 2023-09-22:
I've now tried the code from [this commit](https://github.com/skot/ESP-Miner/commit/66c4b2bb5736b32254846bbba4f32632de7a4ea8) and I get a different result but still it still doesn't want to boot. If it is useful info, I'm using the esp32-s3-wroom-1 and the esp-prog.

[crash.txt](https://github.com/skot/ESP-Miner/files/12699767/crash.txt)

ed0nkey, 2023-09-22:
I may have gotten the 4MB flash esp32 instead of a larger one, is that the issue right? :^)
If so which size should I get?

johnny9, 2023-09-24:
The Bitaxe requires the 16N8R version of the wroom. Check the BOM on the Bitaxe repo for the exact one on DigiKey.

johnny9, 2023-09-24:
That said, I believe IDF should have thrown an error if the flash size was not compatible with your device.

johnny9, 2023-09-24:
Yes, the error in your logfile indicates you have too small of flash to use the normal Bitaxe firmware. You may, however, modify the configuration of the project to not use the ota paritions and instead just have factory and spiffs data. Refer to https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html for how to modify the parition table and use menuconfig to set the flash down to 4MB.

closed 2023-09-24 by johnny9