bitaxeorg/ESP-Miner #43: Not Mining

laststrawman, 2023-10-16:
I am connected to my WiFi but I don't seem to be mining. How do I debug? I am using a Bitaxe board purchased from D-Central. I am trying to connect to public-pool.io

laststrawman, 2023-10-17:
Ok ... update. I flashed the BM1397 following this link: https://github.com/skot/ESP-Miner/releases/tag/v2.0.0
I now can see it talking to public-pool.io but I am still not mining. I definitely have the BM1397 2.2 chip so I set the config.cvs to those settings.

geg780, 2023-10-17:
Do you have a proper power supply? At least 4a are needed.

laststrawman, 2023-10-17:
[image]

laststrawman, 2023-10-17:
I have seen my miner on public-pool.io
[image]

laststrawman, 2023-10-17:
I think you are right ... I looked at my power supply and it has an output of 5v = 1A

geg780, 2023-10-17:
Yeah thats the issue

laststrawman, 2023-10-19:
I am fixed ... it was the power supply.

closed 2023-10-19 by laststrawman