bitaxeorg/ESP-Miner #52: Interrupt startup loop

benjamin-wilson, 2023-10-26:
If the wifi credentials are correct but the pool information is not or the pool is not available/compatible the system will continuously restart.

SACREDHAZE1, 2023-11-23:
HOW TO RESET SETTINGS MANUALY ON HARDWARE MINE HAS THIS ISSUE

benjamin-wilson, 2023-11-23:
> HOW TO RESET SETTINGS MANUALY ON HARDWARE MINE HAS THIS ISSUE

Please refrain from using caps lock... 

Anyway, turn off the wifi router or move it out of range and the bitaxe will create its own SSID again so you can connect directly.

SACREDHAZE1, 2023-11-23:
still has same issue mines 2 accepted and wifi reset

https://github.com/skot/ESP-Miner/issues/52#issuecomment-1824610449

benjamin-wilson, 2023-11-23:
Ok that does not fall under this issue then, please contact the place you purchased it from or OSMU discord help channel for configuration/pool help.

benjamin-wilson, 2023-12-08:
I've run into more situation where this can happen. We need a general way to catch a boot loop.

closed 2024-01-09 by benjamin-wilson