bitaxeorg/ESP-Miner #62: Add support for full stratum URL

Dolu89, 2023-12-03:
Some pools have specific params like nicehash for lower accepted rate
`stratum+tcp://sha256asicboost.auto.nicehash.com:9200#xnsub`
On AxeOS, it's impossible to add this `#xnsub`

source: https://www.nicehash.com/support/mining-help/asic-mining/how-to-connect-your-asic-machine-to-nicehash

benjamin-wilson, 2023-12-25:
I'm not sure how other firmware parse this out but TCP connections can't accept any additional parameters beyond IP and port.

johnny9, 2023-12-25:
Here is the nice hash blog that describes this specific parameter.

https://www.nicehash.com/blog/post/what-is-extranonce-subscribe-extension-xnsub#!

benjamin-wilson, 2023-12-25:
This should be an extra pool option, not in the URL though as the issue describes

Dolu89, 2024-03-02:
XNSUB is now required for mining on nicehash: https://www.nicehash.com/blog/post/notice-for-miners-xnsub-support

CoanLuciano, 2024-03-25:
We could use Bitaxe with Nicehash to make some sats over the lightning network if we solve this issue...