bitaxeorg/ESP-Miner #66: Fan speed 0 rpm with version 2.0.4

CoanLuciano, 2023-12-12:
After upgrade, fan speed always 0rpm. Even after Invert Fan Polarity and disable Automatic Fan Control.

d6n13l0l1v3r, 2023-12-13:
Same issue happen to me with model  BM1366

Power Consumption:	11.66 W
Input Voltage:	5,380 mV
Input Current:	2,169 mA
Frequency:	485 Mhz
Core Voltage:	1200 mV
Measured Core Voltage:	1204 mV
Fan Speed:	0 RPM
Chip Temperature:	46 C

Results

Hash Rate:	472.74 Gh/s
Efficiency:	25.11 W/Th
Best Difficulty:	64.1M

cmelius, 2023-12-15:
I have the same problem with 2.0.4. BM1366
Fan: 0 RPM
Termp: 37.0 C
Pwr: 12,210 W
4847mV: 2530 mA
vCore: 1222 mV

skot, 2023-12-15:
Is the fan actually spinning? What fan do you have installed? How many wires does the fan have?

What did it say the fan speed was before the upgrade?

cmelius, 2023-12-15:
Yes, my fan is spinning. I can't tell which fan is installed currently. I am the guy you retweeted on twitter yesterday with the D-Central Minibit - case. Probably some no name stuff. 
Noctua fan is on it's way to get it more livingroom compatible. Will keep you updated once that is installed. probably tuesday.

d6n13l0l1v3r, 2023-12-15:
answered below:

 iIs the fan actually spinning?    YES
 What fan do you have installed?  capture 1
How many wires does the fan have? 3  capture 2

[image]

[image]

in the setting I have 

Invert Fan Polarity:  checked
Automatic Fan Control: checked 

What did it say the fan speed was before the upgrade? I don't have previous version

CoanLuciano, 2023-12-15:
Actually now I'm running 2.0.1, the fan is spinning and I can read its speed.
[image]

d6n13l0l1v3r, 2023-12-15:
> Actually now I'm running 2.0.1, the fan is spinning and I can read its speed. [...]

what fan do you have ? how many wire?

CoanLuciano, 2023-12-15:
3 wire. Same as  [d6n13l0l1v3r](https://github.com/d6n13l0l1v3r) has

benjamin-wilson, 2023-12-15:
That Chinese fan doesn't actually have a sense wire, the sense wire is connected to ground. The reading is correct.

closed 2023-12-15 by benjamin-wilson

CoanLuciano, 2023-12-18:
So please answer me. Why does it read the fan speed using an old version (2.0.1) Does this read is not real??

cmelius, 2023-12-21:
update: installed Noctua NF-A4x10 PWM and everything works now. 
Fan speed is displayed, RPM is controlled and I am really surprised by the silence. 
I can't tell that the fan is spinning just by the sound of it! The old noname fan was really loud im comparison. 
Highly recommended upgrade!

skot, 2023-12-21:
> So please answer me. Why does it read the fan speed using an old version (2.0.1) The info is not real??

As Ben said, those fans do not have speed sensors. Previous FW versions displayed an incorrect fan speed in these cases.