bitaxeorg/ESP-Miner #67: verbindung bricht ab, factory flash geht nicht

Lollo76, 2023-12-16:
Hello,

I get the normal firmware ﬂashed, but I have the problem that after a few 30 seconds, it writes underwolting that the power supply is the one from bitconmerch and has a constant 5.2xv, I wanted to flash the factory file and it doesn't work because it keeps restarting the flash breaks off after 27%. The problem is that it restarts from the beginning without any settings after 10 minutes or 3 hours.... now nothing shows in the log file in settings... does anyone have an idea?
The model with USB type C FW 2.01 was installed

skot, 2023-12-16:
Make sure you have the barrel power connector inserted all the way.

Lollo76, 2023-12-16:
The plug is in as far as it will go, it just jumps from the picture where it says shares accepted to the wifi
[image]

Lollo76, 2023-12-16:
[image]

benjamin-wilson, 2023-12-18:
You need to fix the restarting issue first, can you post the realtime logs when it happens

Lollo76, 2023-12-18:
hi it showed me this data when booting, then I updated the web.bin, the connection was interrupted due to the undervolting, now I can no longer access the website, but I can under USB ESP tool (website). to access it... the data he showed me...
dx:0,
₿
₿
₿ dx:1 (ifx:0, 48:5d:35:ab:7f:bd), tid:1, ssn:0, winSi
₿
₿
₿ dx:0 (ifx:0, 48:5d:35:ab:7f:bd), tid:0, ssn:58, winSi
₿
₿
₿ dx:0,
₿
₿
₿ dx:1,
₿
₿
₿ dx:0 (ifx:0, 48:5d:35:ab:7f:bd), tid:0, ssn:60, winSi
₿
₿
₿ dx:1 (ifx:0, 48:5d:35:ab:7f:bd), tid:1, ssn:0, winSi
₿
₿
₿ dx:1,
₿
₿
₿ dx:1 (ifx:0, 48:5d:35:ab:7f:bd), tid:1, ssn:0, winSi
₿
₿
₿ dx:1,
₿
₿
₿ dx:1 (ifx:0, 48:5d:35:ab:7f:bd), tid:1, ssn:0, winSi
₿

Lollo76, 2023-12-18:
[image]

benjamin-wilson, 2024-03-15:
Closing inactivity

closed 2024-03-15 by benjamin-wilson