bitaxeorg/ESP-Miner #79: Problems adding and deleting Bitaxes in the swarmfunction

Patsch91, 2024-01-09:
Had 6 Axes integrated in the swarm and deleting one of them from the list resulted in deleting every single one.
Now trying set up a new swarm i cannot add any of them back into it Tried to "start" adding them one by one on each of the OS but everytime i hit the "add" button nothing happens.

all on Version 2.0.5.

I will try if the same behaviour occurs when they are running  on 2.0.4

Patsch91, 2024-01-09:
Just adding another thing that occured couple times before while running on 2.0.4.:  in my case there were three different AxeOS open in one Browser and in one of them i hit the swarmfunction trying to get the three of them into one swarm.  I typed in the IP hit the add button and then the same IP was added three times to the swarm. Maybe this helps improving the swarm function somehow

thalpius, 2024-02-07:
I've had the same issue. I deleted the miners by clicking the remove button, copying the PATCH request and changing the invalid IP address to the IP address of the miner and send it again. Now the list is empty.

Maybe it will help others who wants a clear list before the fix 😊

WantClue, 2024-12-01:
Current swarm does not have any issues anylonger. Swarm has been reworked lately

closed 2024-12-01 by WantClue