bitaxeorg/ESP-Miner #83: Release v2.0.5 froze during www.bin load.  Now the miner won't start up.

scoaste, 2024-01-10:
Bitaxe Ultra miner was at v2.0.4.  Downloaded the esp-miner.bin & www.bin files for v2.0.5 to my local PC.  Went to the setting tab and uploaded the esp-miner.bin file.  Attempted to load the www.bin file.  After 2% it froze and never resume after several minutes.  I attempted to refresh the page and the site was not longer accessible.  I hard-booted the miner and it doesn't come back up.  Now what?

benjamin-wilson, 2024-01-13:
Which version bitaxe do you have, who made it and what version were you updating from?

scoaste, 2024-01-13:
Bitaxe Ultra from D-Central.  Came with v2.0.0.  I upgraded to v2.0.4 with
no issues.  v2.0.5 of esp-miner.bin appeared to upload without issue, but
www.bin froze.  They provided me with a rough guide to flash via bitaxetool
(link below).  I flashed the esp-miner again but don't know what to do with
www.bin which seems to be the problem.

https://d-central.tech/troubleshooting/bitaxe-troubleshooting-guide-solving-operational-challenges/

I translated the above to these:

# install pip, pipx, and bitaxetool (needs its own environment)
sudo aptitude install python3-pip -vy
sudo aptitude install pipx -vy
pipx install bitaxetool

# download bin files, config file example, change to that directory
# modify config
cp config.cvs.example  config.cvs
vi config.cvs

# connect USB, get port, execute flash command
ls -lart /dev/tty*
sudo /home/$USER/.local/bin/bitaxetool -p /dev/ttyACM0 -c ./config.cvs -f
./esp-miner.bin

Regards,
Scott

closed 2024-01-15 by WantClue

scoaste, 2024-01-15:
> Hey Scott,
>
> if your Bitaxe failes during the update process, the partition might be
> bricked and needs to be reflashed using the bitaxetool or the espressif
> software.
> You can also watch videos about how to do this on my channel. Or follow
> the guide in the bitaxe repo.
>
> With this information now, I do close this issue labeled as a question.

As I noted in my previous email, bitaxetool does not solve the problem,
probably because the esp-miner.bin updates without apparent issue, whereas
the www.bin seems to be the cause (i.e
where it froze).

scoaste, 2024-01-15:
This command worked:

sudo /home/$USER/.local/bin/bitaxetool -p /dev/ttyACM0 -c ./config.cvs
-f *./esp-miner-factory-v2.0.6.bin
*