bitaxeorg/ESP-Miner #88: OTA software update gone wrong, bricked miner.

Habies, 2024-01-14:
After a ota update my miner stopped working.
No more display and no wifi.
Is it possible to update the esp32 with a USB to serial interface?
My miner does not have a usb connector.
The board is a JDHX Bit Lottery BM1397 v3.8
Is there a tuturial how to flash the esp32 with a usb-serial converter or a esp-programmer?

skot, 2024-01-14:
You need to reprogram the ESP32-S3 using the ESP-PROG programmer and bitaxetool and/or esptool software. There are several generic ESP32 tutorials out there on this.

unfortunately the manufacturer of your bitaxe has decided to modify the design and not publish documentation on their changes, so you’re on your own there.

WantClue, 2024-01-15:
You can also use A Serial-to-USB Bridge. I did this in one of my first videos about it. It does require a bit of soldering of steady hands with all the pins.
With this information now I close this issue.

closed 2024-01-15 by WantClue