bitaxeorg/ESP-Miner #94: About board version distinction

kakawlala, 2024-01-21:
I got my device from Bitcoin Merch and it has 201 printed on the board.  After updating to version 2.0.7.bin, the power management distinction shows that the board is 204.  I'm really curious if this distinction is correct?

[image]

[image]

Check the 204 board on the Internet. The welding position of ESP32-S3-WROOM-1 should be aligned with the board.
Is my board really 204 instead of 201?

[image]

StarGateMiner, 2024-01-21:
I just updated and my board version says: Board Version 0.11? is this version 201? wonder why its showing in this format.

WantClue, 2024-01-21:
This is something that needs to be adjusted from the manufacture side.
But for the web flasher this is also an issue. I keep this one open until I do have different board version flashs available and keep you posted

closed 2024-03-07 by kakawlala