bitaxeorg/ESP-Miner #95: When the mining pool rejects a submission, the rejected shares are not correctly incremented.

AxisRay, 2024-01-22:
As shown in the picture, my submission was rejected by the pool.
[image]
However, the `result` field in the returned JSON data is `null`.
https://github.com/skot/ESP-Miner/blob/23599cf46f3668af58d252915b347fa14eedd8cf/components/stratum/stratum_api.c#L123-L141
So, in the above code, the result of `if (result_json != NULL && cJSON_IsBool(result_json)) {` is false
Because `null` is not vaild bool value.
Therefore, message->response_success is not being assigned correctly.

skot, 2024-02-16:
Ah, good catch. this would be nice to get fixed so we can see how many shares are _actually_ rejected

benjamin-wilson, 2024-02-25:
Fixed https://github.com/skot/ESP-Miner/commit/feda6609c10b1c37b14fa4f7366f16e4949c30a0

closed 2024-02-25 by benjamin-wilson