bitaxeorg/ESP-Miner #97: System reboots

Sledge0001, 2024-01-28:
System reboots unexpectedly between 4-10 hours using stock settings on firmware v2.0.7 

Issue does not occur on firmware 2.0 4.

yeg0rpetrov, 2024-01-29:
I do have the same issue. But it reboots after 23 hours. Link for context https://damus.io/note1wkgnm62hy55n68h5ccypxajce4z9jvppt3ew4cpj6tcp64s8ffrqsvd5ml

skot, 2024-01-29:
> System reboots unexpectedly between 4-10 hours using stock settings on firmware v2.0.7 
> 
> 
> 
> Issue does not occur on firmware 2.0 4.

Is this with Ocean pool also?

Sledge0001, 2024-01-29:
CK's pool.

Sledge0001, 2024-01-29:
Unfortunately I did not. But it seems to happen only on 2.0.7 it's very stable going back to 2.0.4

yeg0rpetrov, 2024-01-30:
Also just switched from ocean to the https://solo.ckpool.org and it's like restarting every 15 minutes none stop. I tried to re flash the latest firmware again but same issue

yeg0rpetrov, 2024-01-31:
> Also just switched from ocean to the https://solo.ckpool.org and it's like restarting every 15 minutes none stop. I tried to re flash the latest firmware again but same issue 

Update. 14 hours it's running none stop but I already have a first signs that restart is coming. I can't access dashboard via IP. I bought Bitaxe 204 from open source miner com. Attached photo of the board. Maybe will be helpful. [image]

benjamin-wilson, 2024-01-31:
Are you leaving the dashboard open on a computer?

Can you screenshot the whole AxeOS home page please. 
If you can capture the crash in the logs that would also be helpful

yeg0rpetrov, 2024-01-31:
No I don't leave it open. And can't access it right now: says that "Safari can't open the page because the network connection was lost" tried to turn on/off wifi on my phone and laptop same issue.

skot, 2024-01-31:
this sounds like a networking issue maybe? Can you connect to your router and confirm the bitaxe has an IP on your network?

benjamin-wilson, 2024-01-31:
How far away is the bitaxe from your wireless access point?

yeg0rpetrov, 2024-01-31:
Connected devices from router page. espressif is Bitaxe 
All good on my end [image]

Sledge0001, 2024-01-31:
So here's something interesting the 2 xBitAxe's I have from Opensourceminer.com are on 2.0.7 and are stable. The 1 got from altairtech has the issues with 2.0.7

benjamin-wilson, 2024-01-31:
> So here's something interesting the 2 xBitAxe's I have from Opensourceminer.com are on 2.0.7 and are stable. The 1 got from altairtech has the issues with 2.0.7

Well I made them both ;) Can you screenshot the AxeOS dashboard on the one with issues?

Sledge0001, 2024-01-31:
[image]
[image]

Here's the one that's acting odd on 2.0.7 :)

benjamin-wilson, 2024-01-31:
How did you update to 207? Please screenshot AxeOS

Sledge0001, 2024-01-31:
[image]
[image]

Firmware was updated via the AxeOS ui

Sledge0001, 2024-01-31:
I'll try the firmware update again and see if I experience the issue again.

benjamin-wilson, 2024-01-31:
Ok, please screenshot the overview section again when it's updated

Sledge0001, 2024-01-31:
[image]

yeg0rpetrov, 2024-01-31:
I was late. It restarted again while I was figuring out how to use minicom. But seems like I found the error in log [image]

Sledge0001, 2024-01-31:
[image]
And it just crashed.

yeg0rpetrov, 2024-01-31:
> I was late. It restarted again while I was figuring out how to use minicom. But seems like I found the error in log [image]

[image]

benjamin-wilson, 2024-01-31:
What is being used for power supplies?

benjamin-wilson, 2024-01-31:
Oh @Sledge0001 change your vcore to 1200, this is the default. It looks like it changed in the update?

Sledge0001, 2024-01-31:
> Oh @Sledge0001 change your vcore to 1200, this is the default. It looks like it changed in the update?

That's where it's set as per the ui.
[image]

monster4866, 2024-02-01:
u need a better power supply, i have 8 bitaxe, all came with a chep power supply, u need above +5000mv, good 5100-5300mv

https://github.com/skot/bitaxe/issues/118#issuecomment-1904615683

yeg0rpetrov, 2024-02-02:
I heard that I can burn it if more powerful power supply

monster4866, 2024-02-02:
> I heard that I can burn it if more powerful power supply

thats why mr volta invented volt ;)

u should plug only a 5v supply in it

Sledge0001, 2024-02-02:
Over 24 hours online with no rebooting!

yeg0rpetrov, 2024-02-03:
24 hours another one reboot! Ordering this one power supply in hope that gonna fix the issue! https://a.co/d/5srOIj4

I received this one with Bitaxe: https://opensourceminer.com/products/5v-5a-power-supply

yeg0rpetrov, 2024-02-03:
> 24 hours another one reboot! Ordering this one power supply in hope that gonna fix the issue! https://a.co/d/5srOIj4
> 
> I received this one with Bitaxe: https://opensourceminer.com/products/5v-5a-power-supply

I know Benjamin check this thread as well since he was active on here. Your website where I bought the Bitaxe can't be reached! In my email I simply asked to ship back the Bitaxe I bought from you to test it and here what is your website responding to me. [image]

benjamin-wilson, 2024-02-03:
> I know Benjamin check this thread as well since he was active on here. Your website where I bought the Bitaxe can't be reached! In my email I simply asked to ship back the Bitaxe I bought from you [...]

You can email me directly: admin@opensourceminer.com if you want to discuss any issues with your order.

yeg0rpetrov, 2024-02-05:
Was able to cut the log file during restart. Here is log file.
[bitaxe log restarts.txt](https://github.com/skot/ESP-Miner/files/14159978/bitaxe.log.restarts.txt)

yeg0rpetrov, 2024-02-05:
> Was able to cut the log file during restart. Here is log file.
> 
> [bitaxe log restarts.txt](https://github.com/skot/ESP-Miner/files/14159978/bitaxe.log.restarts.txt)
> 
> 

After another 17 hours it has restarted. According to the Ben I have wifi network connection issues. I doubt that this is the case. I did a restart of the modem, talked with the Comcast and everything is healthy. At some point the best difficulty stops going up and it restarts.

benjamin-wilson, 2024-02-05:
I am also investigating a possible memory leak that might be causing this issue.

yeg0rpetrov, 2024-02-06:
> After another 17 hours it has restarted. According to the Ben I have wifi network connection issues. I doubt that this is the case. I did a restart of the modem, talked with the Comcast and [...]

UPDATE! I have tried 2.0.7 and 2.0.6 versions of firmware and device is restarting within ~24 hours. 
After Ben's suggestion to roll back to 2.0.4 device isn't crashing and hashing over 24 hours now. 
The only conclusion what I can make now is that it's not my wifi network connection issue. Might be something with firmware and I am not only the one have this issue. [image][image][image][image][image][image][image][image][image]

benjamin-wilson, 2024-02-06:
[image]
[image]
[image]

2.0.7 Going strong, no memory leaks.

yeg0rpetrov, 2024-02-07:
> @yeg0rpetrov I can't help but notice on your home screen the board version is missing... 
> 
> 
> 
> I'm wondering if with all your troubleshooting something went missing... 
> 
> 
> 
> Possibly flashing your device with some basic manufacturing data to the NVS partition with the correct firmware and a proper config.cvs file might solve your issue. 
> 
> 
> 
> 1. Download Skot ESB-Miner repo from Git to your downloads and unzip it. ./Downloads/ESP-Miner-207
> 
> 2. Download the esp-miner-factory-204-v2.0.7.bin and place in the ESP-Miner folder ./Downloads/ESP-Miner-207/
> 
> 3. Open VS Code, Copy and Paste config.cvs.example and rename the copied file to: config.cvs
> 
> 4. Open config.cvs file and edit the following fields: 
> 
> 
> 
> key,type,encoding,value
> 
> main,namespace,,
> 
> wifissid,data,string,**myssid**
> 
> wifipass,data,string,**mypass** 
> 
> stratumurl,data,string,**solo.ckpool.org**
> 
> stratumport,data,u16,**3333**
> 
> stratumuser,data,string,**mywallet ID+workername**
> 
> stratumpass,data,string,**x**
> 
> asicfrequency,data,u16,**485**
> 
> asicvoltage,data,u16,**1200**
> 
> asicmodel,data,string,**BM1366**
> 
> devicemodel,data,string,**ultra**
> 
> boardversion,data,string,**v2.0.4**
> 
> flipscreen,data,u16,1
> 
> invertfanpol,data,u16,1
> 
> autofanspeed,data,u16,1
> 
> fanspeed,data,u16,100
> 
> --> HIT CTRL-S and SAVE
> 
> 
> 
> 5. Connect your bitaxe
> 
> 6. Select the correct port it connected to (bottom left), select the workspace folder you're working from when prompt 
> 
> 
> 
> 7. Open a terminal and type: bitaxetool --port /dev/cu.usbmodem1101 (or wtv port your Bitaxe is connected to) --config ./config.cvs --firmware ./esp-miner-factory-204-v2.0.7.bin 
> 
> note: Bitaxetool need python version 3.11.7 or earlier, it will not work with the latest python v3.12
> 
> 
> 
> Your bitaxe will flash, reboot and start hashing again. Report back your finding. Take a particular attention the Heap memory: is it steadily going down, or hovering +- a couple 1000 bytes. 
> 
> 
> 
> I think WantClue has good tutorial on hes Youtube channel on how to do this, Bitaxe V2 a new update.
> 
> 
> 
> Hope this help, let us know if you have any questions. :) 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 
> 

The board version probably missing cause I downgraded firmware to the 2.0.4 🤔 Seems like a hell of steps 😄Thank you, will try to do re flash in that way.

benjamin-wilson, 2024-02-07:
> The board version probably missing cause I downgraded firmware to the 2.0.4 🤔 Seems like a hell of steps 😄Thank you, will try to do re flash in that way.

Please don't do that, it's incorrect

yeg0rpetrov, 2024-02-07:
Damn! 1 day and 8 hours it restarted again on 2.0.4 
I give up [image]

benjamin-wilson, 2024-02-07:
Okay it seems conclusive now. This is a network/wireless radio issue. I'm closing the issue.

closed 2024-02-07 by benjamin-wilson