bitaxeorg/ESP-Miner #103: AxeOS nicht erreichbar

thekey82, 2024-02-03:
Nach ca 24-25h ist der Bitaxe nicht mehr über AxeOS erreichbar. Nach einem reboot funktioniert es dann wieder für 24-25h

jddebug, 2024-02-03:
Ich hatte dieses Problem auch. Ich nahm den Kühlkörper ab, ersetzte die Wärmeleitpaste und setzte sie wieder zusammen. Danach waren meine Temperaturen viel niedriger und ich hatte nicht mehr das Problem, den Zugang zu verlieren.

thekey82, 2024-02-04:
An der Temperatur liegt es definitiv nicht , diese liegt bei mir bei 36 Grad.

monster4866, 2024-02-04:
> An der Temperatur liegt es definitiv nicht , diese liegt bei mir bei 36 Grad.

dann hast du v1, da ist der temperatursensor nicht richtig, 36c ist schon etwas hoch, ich habe ca. 30, schick mir mal ein screenshot von den ganzen daten auf der ersten seite links, clock, mv usw.

thekey82, 2024-02-04:
[image]

ich habe 2 Bitaxe. Exakt das selbe Problem bei beiden.

thekey82, 2024-02-04:
Es ist ein 201 Bord

monster4866, 2024-02-04:
> ![camphoto_342241519](https://private-user-images.githubusercontent.com/74416112/302074687-a9840d82-a517-4927-af19-92f52c5f851f.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcwMTcxMDEsIm5iZiI6MTcwNzAxNjgwMSwicGF0aCI6Ii83NDQxNjExMi8zMDIwNzQ2ODctYTk4NDBkODItYTUxNy00OTI3LWFmMTktOTJmNTJjNWY4NTFmLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIwNFQwMzIwMDFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03MmQyOTM5YjFlYWYzOGE2ZGViOWVjNDU4MWM2MGQ5ZjRlZDI2YzFhNDRhNDc5ZTA3MDFjY2U4OTc0NGNiYTQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.K4wYcyS2ymAJgfW6h04ZR1HW30D0iEP3MwT8G2k_omE)
> 
> ich habe 2 Bitaxe. Exakt das selbe Problem bei beiden.

wechsel erstmal das netzteil, wollte nie unter 5000mv fallen

https://amzn.eu/d/1stC5ll

201 ist v1, denke das problem hat sich mit dem neuen netzteil erledigt

monster4866, 2024-02-04:
[image]

so sieht das bei mir aus

thekey82, 2024-02-04:
Okay danke ich werde es mal testen. Wie hoch ist die Temp dann in echt wenn die 36 Grad falsch sind?

benjamin-wilson, 2024-03-15:
Antwort auf den ursprünglichen Kommentar: Ich habe dieses Problem schon einmal gesehen. Es hängt mit etwas in Ihrem Netzwerk zusammen. Wird geschlossen, es sei denn, wir können das Problem reproduzieren.

closed 2024-03-15 by benjamin-wilson