bitaxeorg/ESP-Miner #110: Error when trying to flash the miner

pinokio240, 2024-02-16:
Error when trying to flash the miner
[image]
[image]
If you flash the file esp-miner-factory-v2.0.4.bin without config.cvs using Espressif. The miner starts but cannot be configured. How to flash NVS partition using Espressif? 
Miner: JDHX bit lottery bm1397 v.3.8   z

skot, 2024-02-16:
you can flash just the config.cvs using bitaxetool. `bitaxetool --config ./config.cvs`

glad you got it figured out!

closed 2024-02-16 by skot

johnny9, 2024-02-17:
I removed distutils from bitaxetool. bitaxetool version 0.6 should work with 3.12 on Windows now.

pinokio240, 2024-02-17:
Now the problem is like in https://github.com/skot/ESP-Miner/issues/100. Doesn't measure power correctly and restarts. I tried different firmwares - it didn’t help.

pinokio240, 2024-02-17:
> you can flash just the config.cvs using bitaxetool. `bitaxetool --config ./config.cvs`
> 
> glad you got it figured out!

OK