bitaxeorg/ESP-Miner #113: Firmware Update to 2.0.7 now i can't mine anymore

xXAsystolieXx, 2024-02-23:
Hi everyone, as described in the title, I updated to firmware 2.0.7. Now I only get a 1 with shares accepted but nothing else happens to. I downgrade again to 2.0.6 everything works as it should🤷🏼‍♂️. 

[image]

Low Voltage is shown in the picture because I connected the Bitaxe via USB cable. With the right power cable, this is no longer there. I hope you can help me and I can do the whole thing because I am an absolute beginner and I googled everything I've done so far🤣

xXAsystolieXx, 2024-02-23:
[image]

This is what it looks like with a downgrade after a few seconds

xXAsystolieXx, 2024-02-23:
[image]

And this is what the realtime log looks like on version 2.0.7

xXAsystolieXx, 2024-02-23:
[image]

12 Minuten auf Version 2.0.7  und eine Hashrate von 0

jddebug, 2024-02-23:
Have you tried stock voltage (1200) and stock frequency (485)? I suspect it might work well on 2.0.7 if you do.

xXAsystolieXx, 2024-02-23:
Yes, I've already tried it but without success

jddebug, 2024-02-23:
I have BM1366. It will not work over 60C. Could this be a similar issue with yours?

skot, 2024-02-23:
Which hardware do you have? You say 1366, but the AxeOS says 1397. If that's true it's not going to work.

jddebug, 2024-02-23:
OP said he has the 1397. I have the 1366. I didn't realize the OP had a different hardware than I do.

xXAsystolieXx, 2024-02-23:
I have the BM1397. It has been 60 ° and more since I got the Bitaxe

xXAsystolieXx, 2024-02-23:
Why? Does 2.0.7 not work with the BM1397?

pinokio240, 2024-02-28:
> bm1397

I also have a problem with it (bm1397), now it’s just like that, it’s a waste of money, I tried rolling back to the old firmware to no avail

xXAsystolieXx, 2024-02-28:
B But I don't get 2.0.7 to work ...

pinokio240, 2024-02-28:
@xXAsystolieXx  It falls in love with me, but constantly the nutrition error flies away at the same time as a post -pushing to the network, and nourish from the labator power supply and especially the voltage even up to 5.2 volts lifted it - it did not fall. Not correctly working

xXAsystolieXx, 2024-02-28:
@pinokio240 That sounds like a defective circuit board to me. With 2.0.6 I have no problems with the pool

xXAsystolieXx, 2024-02-28:
@

> @xXAsystolieXx Can you take a picture of your board, and tell us what steps you took to upgrade the firmware?

I can only take a photo later, I'm still on the way. But I connected my board to my PC via USB and with the Bitaxetool command I then flashed the firmware

xXAsystolieXx, 2024-02-28:
key,type,encoding,value
main,namespace,,
asicfrequency,data,u16,475
asicvoltage,data,u16,1400
asicmodel,data,string,BM1397
devicemodel,data,string,max
boardversion,data,string,2.2

These are the parameters I am in theconfig.cvshave deposited

FragOmatig, 2024-02-28:
Hey 👋 
 
 same Problem with my Bitaxe Ultra 1366 😭
Board 204
Firmware v2.0.7 
Chip: BM1366AL 

Voltage is good, no warning. Temp 35 crad.

only 1 share no hash …..
firmware new flashed, but nothing same Problem.
Canot rollback…

xXAsystolieXx, 2024-02-28:
Then I can be happy that I could flash back to 2.0.6 🫤 But good to know that I'm not the only one with this problem

xXAsystolieXx, 2024-02-28:
@qubyt3 I've already tried the web flasher, but this was also unsuccessful😅

pinokio240, 2024-02-28:
> @xXAsystolieXx No problem when you get back, send a picture of your device (front/back).
> 
> Instead of using bitaxetool to update your Bitaxe can you try upgrading with Wantclue new web flasher instead:
> 
> https://wantclue.github.io/bitaxe-web-flasher/. (follow the easy instructions)
> 
> Note: Once you are done, disconnect the usb cable. Go to your wifi networks, connect to Bitaxe_XXXX, and re-enter your wifi ssid/password, stratum url/ port/ user/ password. Click Save then click reboot.

Didn't help https://wantclue.github.io/bitaxe-web-flasher/

FragOmatig, 2024-02-28:
I tried another power supply and web flasher. No way…

pinokio240, 2024-02-28:
[image]
I've finished everything. Probably the power chip has died completely. And what microcircuit should I look for now?

19201003080114, 2024-02-28:
> ![ver_win](https://private-user-images.githubusercontent.com/37944152/308673600-4ab50daa-9428-486a-ad81-34dddf8aa50e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDkxNDYyNTIsIm5iZiI6MTcwOTE0NTk1MiwicGF0aCI6Ii8zNzk0NDE1Mi8zMDg2NzM2MDAtNGFiNTBkYWEtOTQyOC00ODZhLWFkODEtMzRkZGRmOGFhNTBlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjI4VDE4NDU1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRlYmQ1N2RhOGY2ODE0MmUzYjhiZTc0OGRjYjY1ZGE2MjFhYTFiYjdlZDMwMTQ1ODJlOTM3NzQxZjVkYTc4ODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.a3V_WCOaQFGIgH58H8eq6g2Su8DWyljaCrbI2ji8oRI) I've finished everything. Probably the power chip has died completely. And what microcircuit should I look for now? Power supply included

Before you replace anything, try shorting J7 if you have U8 installed. 
[image]

xXAsystolieXx, 2024-02-28:
[image]
[image]

pinokio240, 2024-02-28:
> @pinokio240 В вашем случае, я думаю, вы просто неправильно загрузили прошивку в зависимости от отображаемой версии. Пока не теряйте надежды: P

Honestly, I'm already tired. Now there is another problem: after flashing the miner when the power is turned off, the miner does not start. Shows the network to connect to and the PC does not connect to it.
  Okay, I'll try to look and jump if there is a takach microcircuit

xXAsystolieXx, 2024-02-28:
₿ (93777) stratum_task: rx: {"id":null,"method":"mining.notify","params":["e571c1","30754f29d0feabf790a01894936f5dbcf9eae316000002790000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703b4b30c5075626c69632d506f6f6c","ffffffff02625e362900000000160014ee08418cb52d17f613b6e5466b376d56f8b5c87e0000000000000000266a24aa21a9ed447c55d487ca079f6106d09d44e943bb0d83f4b4590d4ccfaa81c66232db20c000000000",["1f2b5c5747883c21f8e1d12065f3c43f7638a6f1466c8b5b331d5fccf26709e4","f0c39f75b987c8854579d1ca079bbbe24965f1ebd11c6db0bb3801521f815ce5","5a0581d87c3139bc5f08a33aa48555186b4ab1f5698eb4509b9fbd50f0167861","f0fc867115edf5b17127444024e6d4b81f6eb9a245ff38cbf3db3bfdd29321f2","81229b0e848b2f8475806d2eb37d38cf4d3a3678996015229d9ebca97630016c","7a76119b219e4c9361b196023fca93696e0a3c71eca8c5b090efdd2218b19496","9182a2ff8c7d802a3a96b13606bd1c0017735ded457ca4eec5b93b6d541737cd","d27fb4b11c3a7c607548255a301aa75eeb75c9db68d79c660a466587b3e56aa7","65e6a5083eb402e16a9c6d6ea0335ef6e3268bad478481552becfa2d8e9a5cb7","6f14ea48f7ea342d1a80f00c985581613d4968c6101090d9e2b74f6106ce034d","ccd993e781d6b8e4b0749bdfc299468894418ee9d6c1437231820a85d306bdec"],"20000000","170371b1","65df84a1",false]}
₿ (93887) create_jobs_task: New Work Dequeued e571c1
₿ (124197) bm1397Module: return null
₿ (125207) stratum_task: rx: {"id":null,"method":"mining.set_difficulty","params":[8]}
₿ (125217) stratum_task: Set stratum difficulty: 8
₿ (125617) stratum_task: rx: {"id":null,"method":"mining.notify","params":["e575fc","30754f29d0feabf790a01894936f5dbcf9eae316000002790000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703b4b30c5075626c69632d506f6f6c","ffffffff02625e362900000000160014ee08418cb52d17f613b6e5466b376d56f8b5c87e0000000000000000266a24aa21a9ed447c55d487ca079f6106d09d44e943bb0d83f4b4590d4ccfaa81c66232db20c000000000",["1f2b5c5747883c21f8e1d12065f3c43f7638a6f1466c8b5b331d5fccf26709e4","f0c39f75b987c8854579d1ca079bbbe24965f1ebd11c6db0bb3801521f815ce5","5a0581d87c3139bc5f08a33aa48555186b4ab1f5698eb4509b9fbd50f0167861","f0fc867115edf5b17127444024e6d4b81f6eb9a245ff38cbf3db3bfdd29321f2","81229b0e848b2f8475806d2eb37d38cf4d3a3678996015229d9ebca97630016c","7a76119b219e4c9361b196023fca93696e0a3c71eca8c5b090efdd2218b19496","9182a2ff8c7d802a3a96b13606bd1c0017735ded457ca4eec5b93b6d541737cd","d27fb4b11c3a7c607548255a301aa75eeb75c9db68d79c660a466587b3e56aa7","65e6a5083eb402e16a9c6d6ea0335ef6e3268bad478481552becfa2d8e9a5cb7","6f14ea48f7ea342d1a80f00c985581613d4968c6101090d9e2b74f6106ce034d","ccd993e781d6b8e4b0749bdfc299468894418ee9d6c1437231820a85d306bdec"],"20000000","170371b1","65df84a1",false]}
₿ (125727) create_jobs_task: New Work Dequeued e575fc
₿ (125857) bm1397Module: Setting job ASIC mask to 7
₿ (154807) stratum_task: rx: {"id":null,"method":"mining.notify","params":["e57613","30754f29d0feabf790a01894936f5dbcf9eae316000002790000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703b4b30c5075626c69632d506f6f6c","ffffffff02625e362900000000160014ee08418cb52d17f613b6e5466b376d56f8b5c87e0000000000000000266a24aa21a9ed447c55d487ca079f6106d09d44e943bb0d83f4b4590d4ccfaa81c66232db20c000000000",["1f2b5c5747883c21f8e1d12065f3c43f7638a6f1466c8b5b331d5fccf26709e4","f0c39f75b987c8854579d1ca079bbbe24965f1ebd11c6db0bb3801521f815ce5","5a0581d87c3139bc5f08a33aa48555186b4ab1f5698eb4509b9fbd50f0167861","f0fc867115edf5b17127444024e6d4b81f6eb9a245ff38cbf3db3bfdd29321f2","81229b0e848b2f8475806d2eb37d38cf4d3a3678996015229d9ebca97630016c","7a76119b219e4c9361b196023fca93696e0a3c71eca8c5b090efdd2218b19496","9182a2ff8c7d802a3a96b13606bd1c0017735ded457ca4eec5b93b6d541737cd","d27fb4b11c3a7c607548255a301aa75eeb75c9db68d79c660a466587b3e56aa7","65e6a5083eb402e16a9c6d6ea0335ef6e3268bad478481552becfa2d8e9a5cb7","6f14ea48f7ea342d1a80f00c985581613d4968c6101090d9e2b74f6106ce034d","ccd993e781d6b8e4b0749bdfc299468894418ee9d6c1437231820a85d306bdec"],"20000000","170371b1","65df84dd",false]}
₿ (154917) create_jobs_task: New Work Dequeued e57613
₿ (184197) bm1397Module: return null
₿ (185217) stratum_task: rx: {"id":null,"method":"mining.set_difficulty","params":[1]}
₿ (185217) stratum_task: Set stratum difficulty: 1
₿ (185627) stratum_task: rx: {"id":null,"method":"mining.notify","params":["e57a50","30754f29d0feabf790a01894936f5dbcf9eae316000002790000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703b4b30c5075626c69632d506f6f6c","ffffffff02625e362900000000160014ee08418cb52d17f613b6e5466b376d56f8b5c87e0000000000000000266a24aa21a9ed447c55d487ca079f6106d09d44e943bb0d83f4b4590d4ccfaa81c66232db20c000000000",["1f2b5c5747883c21f8e1d12065f3c43f7638a6f1466c8b5b331d5fccf26709e4","f0c39f75b987c8854579d1ca079bbbe24965f1ebd11c6db0bb3801521f815ce5","5a0581d87c3139bc5f08a33aa48555186b4ab1f5698eb4509b9fbd50f0167861","f0fc867115edf5b17127444024e6d4b81f6eb9a245ff38cbf3db3bfdd29321f2","81229b0e848b2f8475806d2eb37d38cf4d3a3678996015229d9ebca97630016c","7a76119b219e4c9361b196023fca93696e0a3c71eca8c5b090efdd2218b19496","9182a2ff8c7d802a3a96b13606bd1c0017735ded457ca4eec5b93b6d541737cd","d27fb4b11c3a7c607548255a301aa75eeb75c9db68d79c660a466587b3e56aa7","65e6a5083eb402e16a9c6d6ea0335ef6e3268bad478481552becfa2d8e9a5cb7","6f14ea48f7ea342d1a80f00c985581613d4968c6101090d9e2b74f6106ce034d","ccd993e781d6b8e4b0749bdfc299468894418ee9d6c1437231820a85d306bdec"],"20000000","170371b1","65df84dd",false]}
[... cut ...] c","7a76119b219e4c9361b196023fca93696e0a3c71eca8c5b090efdd2218b19496","9182a2ff8c7d802a3a96b13606bd1c0017735ded457ca4eec5b93b6d541737cd","d27fb4b11c3a7c607548255a301aa75eeb75c9db68d79c660a466587b3e56aa7","65e6a5083eb402e16a9c6d6ea0335ef6e3268bad478481552becfa2d8e9a5cb7","6f14ea48f7ea342d1a80f00c985581613d4968c6101090d9e2b74f6106ce034d","ccd993e781d6b8e4b0749bdfc299468894418ee9d6c1437231820a85d306bdec"],"20000000","170371b1","65df8519",false]}
₿ (213897) create_jobs_task: New Work Dequeued e57a61

xXAsystolieXx, 2024-02-28:
there is my log with Firmware 2.0.7

xXAsystolieXx, 2024-02-28:
@qubyt3

pinokio240, 2024-02-28:
> @pinokio240 your Bitaxe firmware is not loaded correctly, don't jump anything. Don't for forget you need to hold and press the BOOT button on your Bitaxe, before plugging it in, in order to enable the flash mode. Then connect and follow the rest of the steps.

I don’t have this button, I’m flashing it through the ESPprog programmer. I'll send you photos
[image]
[image]

https://wantclue.github.io/bitaxe-web-flasher/
The device does not respond to reset.
Select your Bitaxe-Model: Bitaxe Max
e: In EFUSE_BLK2__DATA4_REG is used 8 bits starting with 21 bit
D (679) efuse: In EFUSE_BLK2__DATA4_REG is used 3 bits starting with 29 bit
D (686) efuse: In EFUSE_BLK2__DATA5_REG is used 3 bits starting with 0 bit
D (693) efuse: In EFUSE_BLK2__DATA5_REG is used 6 bits starting with 3 bit
D (700) efuse: In EFUSE_BLK2__DATA5_REG is used 6 bits starting with 9 bit
D (707) efuse: In EFUSE_BLK2__DATA4_REG is used 2 bits starting with 0 bit
D (714) efuse: In EFUSE_BLK2__DATA4_REG is used 8 bits starting with 21 bit
D (721) efuse: In EFUSE_BLK2__DATA4_REG is used 3 bits starting with 29 bit
D (728) efuse: In EFUSE_BLK2__DATA5_REG is used 3 bits starting with 0 bit
Terminal disconnected: BufferOverrunError: Buffer overrun

j7 included:

pscreen 2
D (1804) nvs: nvs_close 13
D (1804) nvs: nvs_open_from_partition main 0
D (1804) nvs: nvs_get invertscreen 2
I (1814) wifi station: ESP_WIFI_MODE_STA
I (1814) SystemModule: OLED init success!
D (1824) esp_netif_objects: esp_netif_add_to_list 0x3fcb34f0
D (1824) esp_netif_objects: esp_netif_add_to_list netif added successfully (total netifs: 2)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4005544e  PS      : 0x00060f30  A0      : 0x8203b43f  A1      : 0x3fca8ba0  
A2      : 0x3c0b7250  A3      : 0x00000000  A4      : 0x00000200  A5      : 0x3fc9e574  
Terminal disconnected: BufferOverrunError: Buffer overrun

xXAsystolieXx, 2024-02-28:
@qubyt3 That is the problem, he stays in this loop. The bitaxe does not start to hash

pinokio240, 2024-02-28:
@qubyt3  @19201003080114 

[esp-web-tools-logs.txt](https://github.com/skot/ESP-Miner/files/14439125/esp-web-tools-logs.txt)
This is a log with the latest firmware version v2.0.7. Jumper j7 is enabled, the miner reboots itself

xXAsystolieXx, 2024-02-28:
[image]

pinokio240, 2024-02-28:
> @pinokio240 Wonder why its dropping the socket...hmmm.
> 
> Can you try connecting to solo.ckpool.org port 3333 or public-pool.io port 21496?

sha256.auto.nicehash.com port 9200

xXAsystolieXx, 2024-02-28:
I (27137) wifi:bcn_timeout,ap_probe_send_start
I (36187) wifi:bcn_timeout,ap_probe_send_start
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037b8b2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (27) boot: ESP-IDF v5.1 2nd stage bootloader
I (27) boot: compile time Jan 20 2024 20:00:30
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00400000
I (87) boot:  3 www              Unknown data     01 82 00410000 00300000
I (94) boot:  4 ota_0            OTA app          00 10 00710000 00400000
I (101) boot:  5 ota_1            OTA app          00 11 00b10000 00400000
I (109) boot:  6 otadata          OTA data         01 00 00f10000 00002000
I (117) boot:  7 coredump         Unknown data     01 03 00f12000 00010000
I (124) boot: End of partition table
I (128) boot: Defaulting to factory image
I (133) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=29e94h (171668) map
I (172) esp_image: segment 1: paddr=00039ebc vaddr=3fc98100 size=04bbch ( 19388) load
I (177) esp_image: segment 2: paddr=0003ea80 vaddr=40374000 size=01598h (  5528) load
I (180) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a62e0h (680672) map
I (309) esp_image: segment 4: paddr=000e6308 vaddr=40375598 size=12b54h ( 76628) load
I (335) boot: Loaded app from partition at offset 0x10000
I (335) boot: Disabling RNG early entropy source...
I (346) cpu_start: Multicore app
I (347) cpu_start: Pro cpu up.
I (347) cpu_start: Starting app cpu, entry point is 0x40375584
I (0) cpu_start: App cpu up.
I (365) cpu_start: Pro cpu start user code
I (365) cpu_start: cpu freq: 160000000 Hz
I (365) cpu_start: Application information:
I (368) cpu_start: Project name:     esp-miner
I (373) cpu_start: App version:      v2.0.7
I (378) cpu_start: Compile time:     Jan 20 2024 19:59:47
I (384) cpu_start: ELF file SHA256:  c148b21719898fc7...
I (390) cpu_start: ESP-IDF:          v5.1
I (395) cpu_start: Min chip rev:     v0.0
I (399) cpu_start: Max chip rev:     v0.99 
I (404) cpu_start: Chip rev:         v0.2
I (409) heap_init: Initializing. RAM available for dynamic allocation:
I (416) heap_init: At 3FCA2050 len 000476C0 (285 KiB): DRAM
I (422) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (429) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (435) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (442) spi_flash: detected chip: gd
I (446) spi_flash: flash io: dio
W (450) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (458) sleep: Configure to isolate all GPIO pins in sleep state
I (465) sleep: Enable automatic switching of GPIO sleep configuration
I (472) app_start: Starting scheduler on CPU0
I (477) app_start: Starting scheduler on CPU1
I (477) main_task: Started on CPU0
I (487) main_task: Calling app_main()
I (527) miner: NVS_CONFIG_ASIC_FREQ 475.000000
[... 77 lines cut ...]
I (2367) ASIC_task: ASIC Ready!
I (2377) stratum_task: Socket created, connecting to 68.235.52.36:21496
I (2547) stratum_api: tx: {"id": 1, "method": "mining.subscribe", "params": ["bitaxe/BM1397"]}

I (2787) stratum_api: Received result {"id":1,"error":null,"result":[[["mining.notify","81146af8"]],"81146af8",4]}
I (2787) stratum_api: tx: {"id": 2, "method": "mining.configure", "params": [["version-rolling"], {"version-rolling.mask": "ffffffff"}]}

I (2997) stratum_api: Received result {"id":2,"error":null,"result":{"version-rolling":true,"version-rolling.mask":"1fffe000"}}
I (2997) stratum_api: Set version mask: 1fffe000
I (2997) stratum_api: tx: {"id": 3, "method": "mining.suggest_difficulty", "params": [512]}

I (3017) stratum_api: tx: {"id": 4, "method": "mining.authorize", "params": ["bc1qacyyrr9495tlvyaku4rxkdmd2muttjr73ly6h0.tobisbitaxe", "x"]}

I (3197) stratum_task: rx: {"id":3,"method":"mining.set_difficulty","params":[512]}
I (3197) stratum_task: Set stratum difficulty: 512
I (3507) stratum_task: rx: {"id":4,"error":null,"result":true}
I (3507) stratum_task: message result accepted
I (3917) stratum_task: rx: {"id":null,"method":"mining.notify","params":["eab552","7c498c18c41512ff2535c7fc36607e57fa694e5f0000f6540000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703b9b30c5075626c69632d506f6f6c","ffffffff02b0f99b2800000000160014ee08418cb52d17f613b6e5466b376d56f8b5c87e0000000000000000266a24aa21a9edb5a27171995d5cc96e02fd4612f93797feece2cde4a87b76abbd2f7f870f3d7400000000",["7b705739c2bbe4dd9a2bebc13eb7111eef6c36e1cc38bb0e41c565c75580aeed","17f97459993807e2106d003ad54af8a8726b2166f65a96468542b5707adb757e","ed72583ad42f86613bb01fc950b11d495687579192e227e7d6d136c0932b7ef5","ba6501611d262f405fe75fcc2b728a7cfadf0be0f846e05d3a75e293d196ae6f","7fde5556b4d387441da90d27d8a0179d329974373dfe397451f51dfbee1f1c9b","658adcd0460d5b7f1140725ea05e02e2c2f746cc7025ae778340292974e85168","d4225312a111432 [... cut ...] 68","d4225312a1114322011c82d29e6f10f066d66dd9b53542ec8bd2a676b2398e46","48b80bd64bf15513c5230fb841c6fd7ef139243985872d3f7b9b7392d2223f23","8a0b5771ab41cada8f8e6287b2ee62d6473c21fbfdc7a3378395827bb9c208bf","62ba8b5c8a0a5ad99f07f3ef2d9e42480c955ce3f1e12c911f3a6940b2fe8b07","17d76f604549c537a754016c94fb90dadd632be7176603af55041ea548edb745","ebceb3493b1a2712105e2bd0c8b8e6abfd134b6037443ea51098e5eb97918401"],"20000000","170371b1","65df8d25",false]}
I (32597) create_jobs_task: New Work Dequeued eab55f

xXAsystolieXx, 2024-02-28:
@qubyt3 Is it enough to only connect to the USB cable or should I also connect the power plug?

FragOmatig, 2024-02-28:
my logs on web GUI

webflash logs are clear 

₿ (63612) stratum_task: rx: {"id":null,"method":"mining.set_difficulty","params":[64]}
₿ (63612) stratum_task: Set stratum difficulty: 64
₿ (64132) stratum_task: rx: {"id":null,"method":"mining.notify","params":["e61a60","81ff09bd5762678dd8fbbb63b19e3c2ba14d172200007eed0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703bab30c5075626c69632d506f6f6c","ffffffff02296d7428000000001976a91462e907b15cbf27d5425399ebf6f0fb50ebb88f1888ac0000000000000000266a24aa21a9ed571f858728aa19cb1541efeba867b599c8c3aea6bd4cfeb3b28973cb4b7d30bf00000000",["e5e9a018d52ec5ddd3303cb65fb34a8fd50d214b1944b41b26a2e6c3c369963d","6798da06d42ba6a1fd32774e2b0a62dc7df4ce87a944e99da25b3553398a9a32","c050d791c88d069cd4b2a1e0e4db005f024a0fd8a0f269359bf8356bf81713a0","f6a3fc85052ac2253299c941e1ab11a46fb1d2d2e55cdf85090519431d3d4880","f0b1c51ab5d54f3158f4be46fc4ff059a73c1c058deece94f6c641b7ec31e84f","50a3ca6de90dbc95d1ddc6cd27072a96bb8e3cffc9f90158fc63c91e62991e2e","c3da3f16c0d8b99921af18e636895177af93abed9ad4a77141d029744640b975","ca748e9152b7bf0a7c8cbe36192bcbd4f8316a35df0b5982c976891cb728d7c3","548cf9bbe496db324e53d62c01fbb161c8530fc0015348f63a4a1e4e1063def1","cce0d1bd6465719e3625a8474e47faf341a04d3ef8748dc0a938301e980fadd5","424cf99b1e746552db802f0b94540bf98740565de8b3df52a901bd42fa91c382","cd20662959785da4f29a0d0ac7858dde99732f194c355ed7c0ec5346d4e386b5"],"20000000","170371b1","65df8d57",true]}
₿ (64232) stratum_task: abandoning work
₿ (64232) create_jobs_task: New Work Dequeued e61a60
₿ (123612) stratum_task: rx: {"id":null,"method":"mining.set_difficulty","params":[8]}
₿ (123612) stratum_task: Set stratum difficulty: 8
₿ (124132) stratum_task: rx: {"id":null,"method":"mining.notify","params":["e61a81","81ff09bd5762678dd8fbbb63b19e3c2ba14d172200007eed0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703bab30c5075626c69632d506f6f6c","ffffffff02296d7428000000001976a91462e907b15cbf27d5425399ebf6f0fb50ebb88f1888ac0000000000000000266a24aa21a9ed571f858728aa19cb1541efeba867b599c8c3aea6bd4cfeb3b28973cb4b7d30bf00000000",["e5e9a018d52ec5ddd3303cb65fb34a8fd50d214b1944b41b26a2e6c3c369963d","6798da06d42ba6a1fd32774e2b0a62dc7df4ce87a944e99da25b3553398a9a32","c050d791c88d069cd4b2a1e0e4db005f024a0fd8a0f269359bf8356bf81713a0","f6a3fc85052ac2253299c941e1ab11a46fb1d2d2e55cdf85090519431d3d4880","f0b1c51ab5d54f3158f4be46fc4ff059a73c1c058deece94f6c641b7ec31e84f","50a3ca6de90dbc95d1ddc6cd27072a96bb8e3cffc9f90158fc63c91e62991e2e","c3da3f16c0d8b99921af18e636895177af93abed9ad4a77141d029744640b975","ca748e9152b7bf0a7c8cbe36192bcbd4f8316a35df0b5982c976891cb728d7c3","548cf9bbe496db324e53d62c01fbb161c8530fc0015348f63a4a1e4e1063def1","cce0d1bd6465719e3625a8474e47faf341a04d3ef8748dc0a938301e980fadd5","424cf99b1e746552db802f0b94540bf98740565de8b3df52a901bd42fa91c382","cd20662959785da4f29a0d0ac7858dde99732f194c355ed7c0ec5346d4e386b5"],"20000000","170371b1","65df8d57",true]}
₿ (124242) stratum_task: abandoning work
₿ (124242) create_jobs_task: New Work Dequeued e61a81
₿ (130182) stratum_task: rx: {"id":null,"method":"mining.notify","params":["e61a87","81ff09bd5762678dd8fbbb63b19e3c2ba14d172200007eed0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703bab30c5075626c69632d506f6f6c","ffffffff02296d7428000000001976a91462e907b15cbf27d5425399ebf6f0fb50ebb88f1888ac0000000000000000266a24aa21a9ed571f858728aa19cb1541efeba867b599c8c3aea6bd4cfeb3b28973cb4b7d30bf00000000",["e5e9a018d52ec5ddd3303cb65fb34a8fd50d214b1944b41b26a2e6c3c369963d","6798da06d42ba6a1fd32774e2b0a62dc7df4ce87a944e99da25b3553398a9a32","c050d791c88d069cd4b2a1e0e4db005f024a0fd8a0f269359bf8356bf81713a0","f6a3fc85052ac2253299c941e1ab11a46fb1d2d2e55cdf85090519431d3d4880","f0b1c51ab5d54f3158f4be46fc4ff059a73c1c058deece94f6c641b7ec31e84f","50a3ca6de90dbc95d1ddc6cd27072a96bb8e3cffc9f90158fc63c91e62991e2e","c3da3f16c0d8b99921af18e636895177af93abed9ad4a77141d029744640b975","ca748e9152b7bf0a7c8cbe36192bcbd4f8316a35df0b5982c976891cb728d7c3","548cf9bbe496db324e53d62c01fbb161c8530fc0015348f63a4a1e4e1063def1","cce0d1bd6465719e3625a8474e47faf341a04d3ef8748dc0a938301e980fadd5","424cf99b1e746552db802f0b94540bf98740565de8b3df52a901bd42fa91c382","cd20662959785da4f29a0d0ac7858dde99732f194c355ed7c0ec5346d4e386b5"],"20000000","170371b1","65df8dc5",false]}
₿ (131732) create_jobs_task: New Work Dequeued e61a87
₿ (183692) stratum_task: rx: {"id":null,"method":"mining.set_difficulty","params":[1]}
₿ (183692) stratum_task: Set stratum difficulty: 1
₿ (184112) stratum_task: rx: {"id":null,"method":"mining.notify","params":["e61ecb","81ff09bd5762678dd8fbbb63b19e3c2ba14d172200007eed0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703bab30c5075626c69632d506f6f6c","ffffffff02296d7428000000001976a91462e907b15cbf27d5425399ebf6f0fb50ebb88f1888ac0000000000000000266a24aa21a9ed571f858728aa19cb1541efeba867b599c8c3aea6bd4cfeb3b28973cb4b7d30bf00000000",["e5e9a018d52ec5ddd3303cb65fb34a8fd50d214b1944b41b26a2e6c3c369963d","6798da06d42ba6a1fd32774e2b0a62dc7df4ce87a944e99da25b3553398a9a32","c050d791c88d069cd4b2a1e0e4db005f024a0fd8a0f269359bf8356bf81713a0","f6a3fc85052ac2253299c941e1ab11a46fb1d2d2e55cdf85090519431d3d4880","f0b1c51ab5d54f3158f4be46fc4ff059a73c1c058deece94f6c641b7ec31e84f","50a3ca6de90dbc95d1ddc6cd27072a96bb8e3cffc9f90158fc63c91e62991e2e","c3da3f16c0d8b99921af18e636895177af93abed9ad4a77141d0 [... cut ...] e","c3da3f16c0d8b99921af18e636895177af93abed9ad4a77141d029744640b975","ca748e9152b7bf0a7c8cbe36192bcbd4f8316a35df0b5982c976891cb728d7c3","548cf9bbe496db324e53d62c01fbb161c8530fc0015348f63a4a1e4e1063def1","cce0d1bd6465719e3625a8474e47faf341a04d3ef8748dc0a938301e980fadd5","424cf99b1e746552db802f0b94540bf98740565de8b3df52a901bd42fa91c382","cd20662959785da4f29a0d0ac7858dde99732f194c355ed7c0ec5346d4e386b5"],"20000000","170371b1","65df8e3d",false]}
₿ (251012) create_jobs_task: New Work Dequeued e62319

FragOmatig, 2024-02-28:
@qubyt3

ok thats shit.... i buy the miner with V2.0.7 an not work. 
what can i do?

pinokio240, 2024-02-28:
> @pinokio240 it should be:
> 
> asicfrequency,data,u16,475 asicvoltage,data,u16,1400 asicmodel,data,string,BM1397 devicemodel,data,string,max boardversion,data,string,2.2
> 
> and your showing a firmware version of v.2.0.7 which I think is what you were trying to do :)
but there is more:
key,type,encoding,value
main,namespace,,
........
flipscreen,data,u16,1
invertfanpol,data,u16,1
autofanspeed,data,u16,1
fanspeed,data,u16,100
Are they optional?

xXAsystolieXx, 2024-02-28:
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037b8b2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (27) boot: ESP-IDF v5.1 2nd stage bootloader
I (27) boot: compile time Jan 20 2024 20:00:30
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.2
I (34) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
I (43) boot.esp32s3: SPI Flash Size : 16MB
I (48) boot: Enabling RNG early entropy source...
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (72) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (79) boot:  2 factory          factory app      00 00 00010000 00400000
I (87) boot:  3 www              Unknown data     01 82 00410000 00300000
I (94) boot:  4 ota_0            OTA app          00 10 00710000 00400000
I (101) boot:  5 ota_1            OTA app          00 11 00b10000 00400000
I (109) boot:  6 otadata          OTA data         01 00 00f10000 00002000
I (117) boot:  7 coredump         Unknown data     01 03 00f12000 00010000
I (124) boot: End of partition table
I (128) boot: Defaulting to factory image
I (133) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=29e94h (171668) map
I (172) esp_image: segment 1: paddr=00039ebc vaddr=3fc98100 size=04bbch ( 19388) load
I (177) esp_image: segment 2: paddr=0003ea80 vaddr=40374000 size=01598h (  5528) load
I (180) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a62e0h (680672) map
I (309) esp_image: segment 4: paddr=000e6308 vaddr=40375598 size=12b54h ( 76628) load
I (335) boot: Loaded app from partition at offset 0x10000
I (335) boot: Disabling RNG early entropy source...
I (346) cpu_start: Multicore app
I (347) cpu_start: Pro cpu up.
I (347) cpu_start: Starting app cpu, entry point is 0x40375584
I (0) cpu_start: App cpu up.
I (365) cpu_start: Pro cpu start user code
I (365) cpu_start: cpu freq: 160000000 Hz
I (365) cpu_start: Application information:
I (368) cpu_start: Project name:     esp-miner
I (373) cpu_start: App version:      v2.0.7
I (378) cpu_start: Compile time:     Jan 20 2024 19:59:47
I (384) cpu_start: ELF file SHA256:  c148b21719898fc7...
I (390) cpu_start: ESP-IDF:          v5.1
I (395) cpu_start: Min chip rev:     v0.0
I (399) cpu_start: Max chip rev:     v0.99 
I (404) cpu_start: Chip rev:         v0.2
I (409) heap_init: Initializing. RAM available for dynamic allocation:
I (416) heap_init: At 3FCA2050 len 000476C0 (285 KiB): DRAM
I (422) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (429) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (435) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (442) spi_flash: detected chip: gd
I (446) spi_flash: flash io: dio
W (450) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (458) sleep: Configure to isolate all GPIO pins in sleep state
I (465) sleep: Enable automatic switching of GPIO sleep configuration
I (472) app_start: Starting scheduler on CPU0
I (477) app_start: Starting scheduler on CPU1
I (477) main_task: Started on CPU0
I (487) main_task: Calling app_main()
I (527) miner: NVS_CONFIG_ASIC_FREQ 475.000000
I (527) miner: ASIC: BM1397
I (527) miner: Welcome to the bitaxe!
[... 95 lines cut ...]
I (7867) stratum_task: Set stratum difficulty: 512
I (8067) stratum_task: rx: {"id":4,"error":null,"result":true}
I (8067) stratum_task: message result accepted
I (8477) stratum_task: rx: {"id":null,"method":"mining.notify","params":["eaaef5","dac8a9d1f03f707e706b83c92c51fa6e2eadbaa20001cc4a0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703bcb30c5075626c69632d506f6f6c","ffffffff02f5b2492800000000160014ee08418cb52d17f613b6e5466b376d56f8b5c87e0000000000000000266a24aa21a9eddddf9a631e8f5ffc73dddab53a25cd7cadd09c3fb4753c218f43fbbc284c892f00000000",["81dae03ee49a185cc53e80c9bff8b73b97a6ca5dcdc73d717d319410811df188","4db6b2cecfa1eb71c2e49c05c6f30496c5efc8839c5cb0835e5b15e6d8d96c4e","d3e9ff525ed76a92c0558cb17f88af1067c488412b904b2e70c7f52888e28ed3","8344348f310b3649b42b6617e9d91e5b72dbf51abbf436468e3fc03f9c3cd3c6","20d22ca8ebee49c9673f033ab18d0d3cadcbe1ff43c58803887a015449fab31c","7c661293d9fc8fcc22a24ebe8e9d7c82865b19d41d5e0efc5765939f49584e8c","b6cde26ada3cf8be1a5437183e4dd15ed5733c7de8b8f27cee7fc113c96317e1","ce6bede94f471c9e384ce91b0067ca03f5319f7a93f120776cc52355edda0b58","c906211d3806ed5dbd50aa35e65162b8628d0c0810ce7ce2e53d1f1d5e2d27cc","29f2a776d17746fe18dc0a8708a09c274660fcbc836d48f939a1a95c33d536c1","d6ca2f15e15805d463dbf3cb7501cf12d9adfcefb9f7767cc0fa9f6fafa862de","d135b06f1c215d4dee3dd2d13b14b9d28461447751d5cbc4dc3a3d890e1ff2ed","6d12b155e3e404e18c929c74b55a74997b6adfd76a4f0b51d9a840e39d95c0fc"],"20000000","170371b1","65df9225",false]}
I (8587) SystemModule: Syncing clock
I (8597) create_jobs_task: New Work Dequeued eaaef5
I (8607) bm1397Module: Setting job ASIC mask to 511
I (11037) stratum_task: rx: {"id":null,"method":"mining.notify","params":["eaaeff","dac8a9d1f03f707e706b83c92c51fa6e2eadbaa20001cc4a0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703bcb30c5075626c69632d506f6f6c","ffffffff02f5b2492800000000160014ee08418cb52d17f613b6e5466b376d56f8b5c87e0 [... cut ...] 1","ce6bede94f471c9e384ce91b0067ca03f5319f7a93f120776cc52355edda0b58","c906211d3806ed5dbd50aa35e65162b8628d0c0810ce7ce2e53d1f1d5e2d27cc","29f2a776d17746fe18dc0a8708a09c274660fcbc836d48f939a1a95c33d536c1","d6ca2f15e15805d463dbf3cb7501cf12d9adfcefb9f7767cc0fa9f6fafa862de","d135b06f1c215d4dee3dd2d13b14b9d28461447751d5cbc4dc3a3d890e1ff2ed","6d12b155e3e404e18c929c74b55a74997b6adfd76a4f0b51d9a840e39d95c0fc"],"20000000","170371b1","65df92d9",false]}
I (131787) create_jobs_task: New Work Dequeued eab7ef

xXAsystolieXx, 2024-02-28:
now with power supply ;)

xXAsystolieXx, 2024-02-28:
@qubyt3  No, unfortunately not

pinokio240, 2024-02-28:
[image]
An endless reboot begins
[esp-web-tools-logs.txt](https://github.com/skot/ESP-Miner/files/14439586/esp-web-tools-logs.txt)
Goodbye, I'm off to bed, it's night time

xXAsystolieXx, 2024-02-28:
@qubyt3 No, didn't overclocked him. The strange thing is, if I flashing back to version 2.0.6 now, my bitaxe starts to hash🤣

benjamin-wilson, 2024-02-28:
@pinokio240 did yours ever work? You're missing 6 resistors at the top.

benjamin-wilson, 2024-02-28:
Ah that was a cheap one off AliExpress. I'm not surprised. Buy them off this list for high quality units with retailers that will stand behind the product and offer a warranty. 
https://bitaxe.org/legit.html

xXAsystolieXx, 2024-02-28:
I (178360) asic_result: Nonce difficulty 580.56 of 1000.ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037b8a2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16f0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (27) boot: ESP-IDF v5.1-564-g8d2dbd461f 2nd stage bootloader
I (27) boot: compile time Jan 12 2024 14:00:52
I (27) boot: Multicore bootloader
I (31) boot: chip revision: v0.2
I (35) boot.esp32s3: Boot SPI Speed : 80MHz
I (40) boot.esp32s3: SPI Mode       : DIO
I (44) boot.esp32s3: SPI Flash Size : 16MB
I (49) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (73) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (80) boot:  2 factory          factory app      00 00 00010000 00400000
I (88) boot:  3 www              Unknown data     01 82 00410000 00300000
I (95) boot:  4 ota_0            OTA app          00 10 00710000 00400000
I (103) boot:  5 ota_1            OTA app          00 11 00b10000 00400000
I (110) boot:  6 otadata          OTA data         01 00 00f10000 00002000
I (118) boot:  7 coredump         Unknown data     01 03 00f12000 00010000
I (126) boot: End of partition table
I (130) boot: Defaulting to factory image
I (134) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=29ee0h (171744) map
I (174) esp_image: segment 1: paddr=00039f08 vaddr=3fc98400 size=04ba4h ( 19364) load
I (178) esp_image: segment 2: paddr=0003eab4 vaddr=40374000 size=01564h (  5476) load
I (181) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a6320h (680736) map
I (310) esp_image: segment 4: paddr=000e6348 vaddr=40375564 size=12e00h ( 77312) load
I (336) boot: Loaded app from partition at offset 0x10000
I (337) boot: Disabling RNG early entropy source...
I (348) cpu_start: Multicore app
I (348) cpu_start: Pro cpu up.
I (348) cpu_start: Starting app cpu, entry point is 0x4037558c
I (0) cpu_start: App cpu up.
I (366) cpu_start: Pro cpu start user code
I (366) cpu_start: cpu freq: 160000000 Hz
I (367) cpu_start: Application information:
I (369) cpu_start: Project name:     esp-miner
I (375) cpu_start: App version:      v2.0.6
I (379) cpu_start: Compile time:     Jan 12 2024 14:00:45
I (386) cpu_start: ELF file SHA256:  d01bbe994f06fd40...
I (391) cpu_start: ESP-IDF:          v5.1-564-g8d2dbd461f
I (398) cpu_start: Min chip rev:     v0.0
I (402) cpu_start: Max chip rev:     v0.99 
I (407) cpu_start: Chip rev:         v0.2
I (412) heap_init: Initializing. RAM available for dynamic allocation:
I (419) heap_init: At 3FCA2350 len 000473C0 (284 KiB): DRAM
I (425) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (432) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (438) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (445) spi_flash: detected chip: gd
I (449) spi_flash: flash io: dio
W (453) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
I (461) sleep: Configure to isolate all GPIO pins in sleep state
I (468) sleep: Enable automatic switching of GPIO sleep configuration
I (475) app_start: Starting scheduler on CPU0
I (480) app_start: Starting scheduler on CPU1
I (480) main_task: Started on CPU0
I (490) main_task: Calling app_main()
I (530) miner: NVS_CONFIG_ASIC_FREQ 475.000000
I (530) miner: ASIC: BM1397
I (530) miner: Welcome to the bitaxe!
[... 329 lines cut ...]
I (158440) stratum_task: rx: {"id":47,"error":null,"result":true}
I (158440) stratum_task: message result accepted
I (160160) asic_result: Nonce difficulty 13950.16 of 512.
I (160190) SystemModule: Network diff: 81725299822043.218750
I (160190) stratum_api: tx: {"id": 48, "method": "mining.submit", "params": ["bc1qacyyrr9495tlvyaku4rxkdmd2muttjr73ly6h0.tobisbitaxe", "e6f2a5", "4c0c0000", "65df9978", "fa448fcf", "00006000"]}

I (160390) stratum_task: rx: {"id":48,"error":null,"result":true}
I (160390) stratum_task: message result accepted
I (167170) asic_result: Nonce difficulty 1590.99 of 512.
I (167170) stratum_api: tx: {"id": 49, "method": "mining.submit", "params": ["bc1qacyyrr9495tlvyaku4rxkdmd2muttjr73ly6h0.tobisbitaxe", "e6f2a5", "090f0000", "65df9978", "368b474c", "00000000"]}

I (167450) stratum_task: rx: {"id":49,"error":null,"result":true}
I (167450) stratum_task: message result accepted
I (171380) asic_result: Nonce difficulty 1069.36 of 512.
I (171380) stratum_api: tx: {"id": 50, "method": "mining.submit", "params": ["bc1qacyyrr9495tlvyaku4rxkdmd2muttjr73ly6h0.tobisbitaxe", "e6f2a5", "ae100000", "65df9978", "4d89440e", "00004000"]}

I (171590) asic_result: Nonce difficulty 3254.02 of 512.
I (171590) stratum_api: tx: {"id": 51, "method": "mining.submit", "params": ["bc1qacyyrr9495tlvyaku4rxkdmd2muttjr73ly6h0.tobisbitaxe", "e6f2a5", "c3100000", "65df9978", "1c040b4f", "00006000"]}

I (171750) stratum_task: rx: {"id":50,"error":null,"result":true}
I (171750) stratum_task: message result accepted
I (171960) stratum_task: rx: {"id":51,"error":null,"result":true}
I (171960) stratum_task: message result accepted
I (176640) asic_result: Nonce difficulty 1343.47 of 512.
I (176640) stratum_api: tx: {"id": 52, "method": "mining.submit", "params": ["bc1qacyyrr9495tlvyaku4rxkdmd2muttjr73ly6h0.tobisbitaxe", "e6f2a5", "bc120000", "65df9978", "35b64d32", "00000000"]}

xXAsystolieXx, 2024-02-28:
Only for comparison, this log is with version 2.0.6 and my Bitaxe starts to Hashen

FragOmatig, 2024-02-28:
> 
> Your board version is wrong, it should be 0.11 by the way, not "204". Seems you may have changed that in the config.cvs file by accident.

can anyone help me plz?

xXAsystolieXx, 2024-02-28:
> > Your board version is wrong, it should be 0.11 by the way, not "204". Seems you may have changed that in the config.cvs file by accident.
> 
> can anyone help me plz

How do you flash your bitaxe?

FragOmatig, 2024-02-28:
with https://wantclue.github.io/bitaxe-web-flasher/ 

i dont now what the last user do with it.

FragOmatig, 2024-02-28:
thx i try it :)

xXAsystolieXx, 2024-02-28:
@FragOmatig Have you ever tried the terminal in Linux?

FragOmatig, 2024-02-28:
no i have a laptop with ubuntu, thats a good idea.

xXAsystolieXx, 2024-02-28:
I just don't know if Windows alone works with the PowerShell. I just don't know why you should have board version 0.11 instead of 204, I can't get a version number on your board

FragOmatig, 2024-02-28:
thats the riddle. 
on ubuntu i must install bitaxetool for the console?

xXAsystolieXx, 2024-02-28:
> no i have a laptop with ubuntu, thats a good idea.

Download the .zip with the ESP Master file under <code>  Then install Bitaxetool. Then you edit yours as in the instructions on this Github page from Skotconfig.cvsAnd save them in the ESP master folder. Connect                                                                                                                                                                                 Then you save the desired factory file in the ESP Master folder. Connect Bitaxe The Bitaxe Copy command from the page here and enter your factory version.Now just push the enter button and it works

xXAsystolieXx, 2024-02-28:
What happened to the formation🙈

xXAsystolieXx, 2024-02-28:
> thats the riddle. on ubuntu i must install bitaxetool for the console?

Yes

xXAsystolieXx, 2024-02-28:
@qubyt3  Ok, I understand I'm trying to explain it to him as I did it. Maybe I can help him like that👌🏻

FragOmatig, 2024-02-28:
i try esp-miner.bin and www.bin in web Gui. On screen Show Working for any minutes an close. 
no error or succsess text are showing. not mining. F***ing up :D  

lets try ubuntu console. installing python3-pip....

[32 more replies]

closed 2024-03-15 by benjamin-wilson