bitaxeorg/ESP-Miner #122: Bug: Global Restart Button is not visible on Mobile Phone Browsing

TheMissingNTLDR, 2024-03-05:
Steps:
1. Using BitAxe with BM1366 here
2. Bug applies to this version only [v2.1.0](https://github.com/skot/ESP-Miner/releases/tag/v2.1.0)
3. download and upgrade to [esp-miner.bin](https://github.com/skot/ESP-Miner/releases/download/v2.1.0/esp-miner.bin) [www.bin](https://github.com/skot/ESP-Miner/releases/download/v2.1.0/www.bin)
4. Open the www website on a mobile phone via the IP address
5. I am using iPhoneX and Safari App as browser
6. Actual Result: User is not able to see the (Global) Restart Button
7. Expected Result: User should be able to see the (Global) Restart Button

WORKAROUND: Add your IP to Swarm. Go to Swarm and use the little square Restart button shown there within your worker row.

skot, 2024-03-09:
I'm seeing the same here. restart button is nowhere to be found on iOS Safari. 
[image]

Sledge0001, 2024-03-10:
Same on Android!

benjamin-wilson, 2024-03-15:
Fixed eb55394d0afe550f5646d5f58a2b118b8a399aeb

closed 2024-03-15 by benjamin-wilson