bitaxeorg/ESP-Miner #128: V2.1.0 GUI improvements

19201003080114, 2024-03-10:
1. Rename "ASIC **Voltage** Requested" to "ASIC **Current** Requested"

2. Add Fan RPM together with Fan %

3. Fan % is always at 88% while I can hear the fan speed is being adjusted by software. Perhaps a bug?

[image]

TheRockaXXX, 2024-03-10:
I can confirm, that the percentage of the fan speed stucks.

19201003080114, 2024-03-12:
Renamed subject to V2.1.1 as the same GUI improvements / bugs still apply

skot, 2024-03-12:
1. It should still be **voltage** requested. I think the units just need to change

benjamin-wilson, 2024-03-15:
Fixed the typo 4cb30a19a46fbefbdf52925e23244654979138c9

19201003080114, 2024-03-15:
I sketched some suggestions for GUI improvements

[image]

thebullishbitcoiner, 2024-03-15:
Is it possible to get some/all of the old power meters into the new GUI? I think the only one missing might be input current. That measurement really helped me figure out why my Bitaxe kept showing as offline in my Braiins dashboard.

I kept having to restart the Bitaxe for it to appear again (and then go offline moments later). I figured out that I didn't plug the power cable in all the way. It's been running stable (i.e. ~2200 mA) ever since.

[image]

closed 2024-04-05 by skot