bitaxeorg/ESP-Miner #137: Time management for frequency and core voltage

exobug, 2024-03-15:
Is it possible to implement a menu for time management, e.g. to use a special frequency and core voltage at night from 11 p.m. to 8 a.m.?

And during the rest of the time for the default settings? Is that possible?

19201003080114, 2024-03-15:
You can use a keyboard recorder to record the keystrokes that will go to the webpage and change the settings. Then run the macro using a scheduler.

exobug, 2024-03-16:
> You can use a keyboard recorder to record the keystrokes that will go to the webpage and change the settings. Then run the macro using a scheduler.

but there is a Scheduler in Bitaxe firmware and time can be updated over internet. nor?

skot, 2024-03-16:
The ESP32 has a RTC that could be used for this, so yes technically possible. It's a big task to implement in firmware and the UI.

I think a better way to do it might be a separate program running on another machine to control the miners over the API.

Please use the Discussion area or OSMU Discord to discuss feature requests.

closed 2024-03-16 by skot

rdmxq, 2025-09-17:
It would be a great feature.