bitaxeorg/ESP-Miner #138: Bitaxe failed to recieved New-Stratum-Task-Work after WAN-IP-Change (no WAN-IP for e.g. 120s) - Bug?

seepv, 2024-03-16:
If the router get's a new IP from a Internet-Provider the Bitaxe failed to reconect for new work.

The same ocured after an OpenWRT-WAN-Network-Restart.

Is this a AxeOS or a Firmware Bug?

Is there e.g. a watchdog somewere (in AxeOS or Firmware) that resets the Bitaxe if recieving or submitting found Blocks failed for a certain time?

Bitaxe 204; AxeOS 2.1.1

seepv, 2024-03-17:
@skot - If for e.g. the WAN-IP changes the mining subscribtion expires.
Without a new mining subscribtion, the Bitaxe keeps 'hashing' in a loop with old data.
That's what's going on - or am I wrong?

skot, 2024-03-17:
The firmware can generate a lot of work by rolling extranonce. But we should notice the server is no longer connected pretty soon when attempting to submit a share.

closed 2024-04-05 by skot