bitaxeorg/ESP-Miner #145: SELF TEST POWER: FAIL with v2.1.2 and v2.1.3 and BItAxe Supra PCB Rev 401

hanleyp, 2024-03-21:
SELF TEST POWER: FAIL with v2.1.2 and v2.1.3 and BItAxe Supra PCB Rev 401

**To Reproduce**
Steps to reproduce the behavior:
1. Upgrade to v2.1.2 or v2.1.3 using asp-miner.bin via web or esp-miner-factory-401-v2.1.2/3 using web.
2. Reboot and see failure.
This always occurs for me with v2.1.2 and v2.1.3. v2.1.1 is working (no SELF TEST). But I had to restore with esp-miner-factory-205-v2.1.1.bin.

**Expected behavior**
SELF TEST POWER: PASS

**Hardware (please complete the following information):**
 - Bitaxe HW version: Supra 401
 - Bitaxe HW vendor: altairtech.io
 - ESP-Miner FW version: 2.1.3 and 2.1.3
 - Voltage:1.2V

hanleyp, 2024-03-21:
Looks like 'power_consumption_pass' is set to < 13, but I'm getting higher. So far, I haven't been able to get it to PASS by setting config.cvs to 400 & 1100. Maybe it tests before applying those settings.

closed 2024-04-05 by skot

Shen, 2024-05-21:
Why is this issue closed? It's still not working.

skot, 2024-05-21:
> Why is this issue closed? It's still not working.

What's not working for you, specifically?

Shen, 2024-05-21:
> What's not working for you, specifically?

SELF TEST POWER: FAIL
Bitaxe HW version: Supra 400
Bitaxe HW vendor: d-central.tech
ESP-Miner FW version: 2.1.3 (with https://wantclue.github.io/bitaxe-web-flasher/)
Voltage:1.2V

With V 2.1.1 the system boots regulary.

skot, 2024-05-21:
Are you using the "factory" image? This is only for first-time programming (ie at a factory). Please try the "just update" image if you are not already.

Shen, 2024-05-21:
> Are you using the "factory" image? This is only for first-time programming (ie at a factory). Please try the "just update" image if you are not already.

I tried. SELF TEST POWER: FAIL

skot, 2024-05-21:
Ah, ok I think we have a confusing setup here. I'll release a new esp-miner with this cleared up, hopefully. https://github.com/skot/ESP-Miner/issues/177

in the mean time can you use v2.1.1?

Shen, 2024-05-21:
Yes, thank you!
I will try out the new version when it is ready.