bitaxeorg/ESP-Miner #146: Feature request: Optimise GUI and expand SWARM overview

zcht, 2024-03-21:
The current update looks really good, even if I don't like the somewhat inaccurate vCore display. An option of both would be very helpful here, a graphical one but also in text form like before the GUI update

It would be desirable if we could get the GUI to work like this: https://github.com/shufps/piaxe-miner

Especially if several bitaxes are operated in the swarm, such an overview in the swarm would be very helpful. Furthermore, the graphs should save the values when a bitaxe is restarted. Unfortunately, this is not the case at the moment.

Many thanks 4 the OpenSource project.

lv333ming, 2024-04-02:
I hope there can be a visual interface for parameter configuration, which will be much more efficient. The control threshold is also low.

WantClue, 2024-12-01:
GUI has been reworked is more accurate and swarm is operational normal again

closed 2024-12-01 by WantClue