bitaxeorg/ESP-Miner #160: Under version V2.1.3, the best difficulty cannot be displayed properly.

kakawlala, 2024-04-06:
I am under version V2.1.3 and got 5.78G again today.  It can display normally at the moment, but after restarting, it still records the upper limit of 4.29G.  It seems it's still unresolved.

[image]

[image]

skot, 2024-04-06:
Okay, it looks like we're truncating the saved nvm value. Thanks for reporting this!

benjamin-wilson, 2024-04-07:
This was fixed a few days ago, needs to be published.
https://github.com/skot/ESP-Miner/commit/6283a68d4ba7cf406ad570917a62a5bdd47b1395

closed 2024-04-07 by benjamin-wilson