bitaxeorg/ESP-Miner #161: LuckyMiner LV06 is not supported

PSLLSP, 2024-04-07:
It would be nice to have support for Lucky Miner LV06. The first step should be to find differences in hardware design. It is a "simple" device, just some ESP32, ASIC chip BM1366, LCD and PM chip. Do I miss something? I assume that only few IO pins are connected in different way to ASIC chip.

I assumed they cloned BitAxe but maybe they did not cloned it, or they cloned it in a way that it is not 100% compatible... It would be nice to know details.

There are some rumors that flashing LV06 with ESP-Miner firmware bricks the device, that it cannot get IP address and it is "bricked" because Web interface cannot be accessed (no IP address).

benjamin-wilson, 2024-04-07:
LV06 is a closed source clone, violating the licenses we provided, we will not support it.

closed 2024-04-07 by benjamin-wilson

PSLLSP, 2024-04-07:
Many open source projects work in just opposite way. They design alternative firmware, to replace closed source firmware... You stated, that you are not interested in this and it looks like you even added some test that prevents to run BitAxe firmware on LuckyMiner LV06, and it bricks the device. Or was that just a side effect?