bitaxeorg/ESP-Miner #172: Reworking the auto fan functionality

WantClue, 2024-04-30:
This is just an open issue to keep track of a potential rework for the auto fan functionality. Despite the fact that it works, it does tune the fan way too often and way too harsh. A slighter approach is needed. Maybe stages of ramp up and down.

tdb3, 2024-06-04:
Was this addressed since this Issue was opened?

If not, it seems like this could be somewhat straightforward tweak to `automatic_fan_speed()`.

https://github.com/skot/ESP-Miner/blob/38fb4ae999c039ac5bdd5adf8267d94e98442192/main/tasks/power_management_task.c#L189-L208

Looks like fan speed is adjusted every 2s, and done so with a smooth adjustment rather than a stepped one.

https://github.com/skot/ESP-Miner/blob/38fb4ae999c039ac5bdd5adf8267d94e98442192/main/tasks/power_management_task.c#L185C9-L185C28

Here are some initial/thoughts/alternatives:
 - Option 1: Keep fan speed adjustment as-is (smooth adjustment once every 2s based on chip temperature).  Slight pitch variation may be heard as the fan speed is balances.
 - Option 2: Use a stepped fan speed with some hysteresis around step points.  Chip temperature is allowed to vary more than option 1.  Fan speed adjusts in stepped levels, which will have different noise/pitch characteristics.  Changes from one step to another is likely more noticeable than option 1.
 - Option 3:  Introduce more lag in fan speed adjustment (slower ramp up and slower ramp down), almost like fan speed based on a moving average of chip temperature.  Chip temperature is allowed to vary more than option 1 but less than option 2.

skot, 2024-10-10:
There has been some talk of implementing PID fan control..

WantClue, 2025-04-16:
done with PID controller