bitaxeorg/ESP-Miner #176: Add `password` field on Settings page (for `mining.authorize`)

plebhash, 2024-05-21:
The AxeOS UI only allows me to type in a `username`, but not a `password`.

I want to do some [Solo SV1 mining on Demand pool](https://www.dmnd.work/#solo-sv1), which requires the `password` field for authentication:

[image]

Fi3, 2024-05-21:
I confirm that in order to mine in solo mode with sv1 with demand you need to be able to insert a password.

benjamin-wilson, 2024-05-24:
Maybe I'm missing something but there is already an input for a stratum password in settings.
[image]

closed 2024-05-26 by benjamin-wilson

plebhash, 2024-05-26:
my BitAxe firmware is probably out of date, sorry for the noise

plebhash, 2024-05-29:
@benjamin-wilson how do I download or generate a `www.bin` so I can update my UI?

benjamin-wilson, 2024-05-29:
There's also the www.bin on the releases page. If there's no UI changes I don't upload one.