bitaxeorg/ESP-Miner #178: Won't boot, screen blank after update to 2.14

jddebug, 2024-05-22:
Won't boot, screen blank after update to 2.14

Updated 4 others with no problem. Any pointers on how to recover?

skot, 2024-05-22:
are these updates through AxeOS? What bitaxe hw?

you might try using the bitaxe web flasher to go back and try again; https://wantclue.github.io/bitaxe-web-flasher/

jddebug, 2024-05-22:
Yes using AxeOS. Board version 0.11

I will take a look at that link.

jddebug, 2024-05-22:
ok, I installed the bitaxe-flasher. It appears I still have to do something after that to get the firmware installed? Flasher install was successful but not booting up as a miner. Is there a next step?

jddebug, 2024-05-22:
Nevermind. Did it twice and now all good. Thanks

jddebug, 2024-05-23:
Ok, it boots and looks like the web interface I am used to but it thinks the ASIC is at 127.75C and will not mine. I have the board version 0.11 and was not sure which bite model to choose from the list so I went with Bite Max since it didn't have a board version listed. I don't think it is setting things up correctly. This miner has been flawless until trying to update to 2.14.

Any guidance?

closed 2024-05-23 by skot