bitaxeorg/ESP-Miner #184: Reconnect to Wifi works but doesn't start mining 2.1.5

jddebug, 2024-05-25:
**Describe the bug**
Router rebooted today. Miners lost connection but seem to have all (5 of them) connected back to wifi after it came back online. However, all 5 did not start mining. 

I can connect to them on the local LAN. This is what I am seeing in the logs:

₿ (899188) stratum_task: Socket unable to connect to public-pool.io:21496 (errno 113)
₿ (904188) stratum_task: Socket created, connecting to 0.0.0.0:21496

One seems to have rebooted on its own after about 18 minutes and it is now mining. The others have not rebooted and are still showing that same information in the logs. Rebooting them gets them mining again.

mac54, 2024-05-31:
I have the same problem with 2.1.6. Wifi reconnect works and webinterface is reachable again in LAN. But mining does not start.

alejandroperezlopez, 2024-05-31:
I've tested it some minutes ago and it's working on my two Bitaxe 204 (Ultra). When the wifi network was down, both devices were showing both connecting to my ssid and the AP ssid available messages.

When the network was available again, both connected and started showing the mining information.

mac54, 2024-06-01:
For me it's not working:
₿ (368759) stratum_task: Socket unable to connect to solo.ckpool.org:3333 (errno 113)
₿ (373759) stratum_task: Socket created, connecting to 0.0.0.0:3333
₿ (392259) stratum_task: Socket unable to connect to solo.ckpool.org:3333 (errno 113)
₿ (397259) stratum_task: Socket created, connecting to 0.0.0.0:3333
₿ (415759) stratum_task: Socket unable to connect to solo.ckpool.org:3333 (errno 113)
₿ (420759) stratum_task: Socket created, connecting to 0.0.0.0:3333
₿ (439259) stratum_task: Socket unable to connect to solo.ckpool.org:3333 (errno 113)
₿ (444259) stratum_task: Socket created, connecting to 0.0.0.0:3333
₿ (462759) stratum_task: Socket unable to connect to solo.ckpool.org:3333 (errno 113)
₿ (467759) stratum_task: Socket created, connecting to 0.0.0.0:3333
₿ (486259) stratum_task: Socket unable to connect to solo.ckpool.org:3333 (errno 113)
₿ (491259) stratum_task: Socket created, connecting to 0.0.0.0:3333

skot, 2024-06-01:
0.0.0.0 sounds like a DNS problem somewhere.

solobito, 2024-06-01:
I've seen this too. It's possible that once the connection is lost, it can't resolve the URL to an IP address.

solo.ckpool.org is 2604:2dc0:100:240f::1

Maybe we could try mining directly with the pool's IP address to see if there are any errors.

pixeldoc2000, 2024-06-05:
Same Problem with 2.1.6 and 2.17 and possible older versions. 

```
₿ (18824034) stratum_task: Socket unable to connect to pool.example.org:3333 (errno 113)
₿ (18829034) stratum_task: Socket created, connecting to 0.0.0.0:3333
```

Looking at the source, it seems like stratum_task() does not check if the dns lookup was successful (resolved ip = 0.0.0.0) and does not try to resolve the dns again.

Then stratum_task() does try to connect to the unresolved ip endlessly -> can't connect to pool.

skot, 2024-06-05:
when a pool URL resolves to 0.0.0.0 that likely means there was a DNS failure. We should be recognizing this error and retrying until it works.

skot, 2024-06-05:
Ah, I see what's going on here. The DNS lookup is not in the retry loop. So if for whatever reason DNS fails it sets the IP to 0.0.0.0. The Bitaxe will just keep retrying this bad IP.

```
I (9701) stratum_task: Get IP for URL: publisdfsdfsfc-pool.io

I (9721) stratum_task: Connecting to: stratum+tcp://publisdfsdfsfc-pool.io:21496 (0.0.0.0)

I (9721) stratum_task: Socket created, connecting to 0.0.0.0:21496
I (9731) ASIC_task: ASIC Ready!
I (9731) main_task: Returned from app_main()
E (27971) stratum_task: Socket unable to connect to publisdfsdfsfc-pool.io:21496 (errno 113)
I (32971) stratum_task: Socket created, connecting to 0.0.0.0:21496
E (51471) stratum_task: Socket unable to connect to publisdfsdfsfc-pool.io:21496 (errno 113)
I (56471) stratum_task: Socket created, connecting to 0.0.0.0:21496
E (74971) stratum_task: Socket unable to connect to publisdfsdfsfc-pool.io:21496 (errno 113)
I (79971) stratum_task: Socket created, connecting to 0.0.0.0:21496
E (98471) stratum_task: Socket unable to connect to publisdfsdfsfc-pool.io:21496 (errno 113)
I (103471) stratum_task: Socket created, connecting to 0.0.0.0:21496
E (121971) stratum_task: Socket unable to connect to publisdfsdfsfc-pool.io:21496 (errno 113)
I (126971) stratum_task: Socket created, connecting to 0.0.0.0:21496
```

skot, 2024-06-05:
I made a 2.1.8b1 build with this fix; can y'all give it a try? 
You can unzip this image and then update via AxeOS. No need to worry about www.bin if you're already on 2.1.7
[esp-miner.bin.zip](https://github.com/user-attachments/files/15596411/esp-miner.bin.zip)

pixeldoc2000, 2024-06-08:
With 2.1.8 i had at least one instance where dns lookup only partialy resolved the pool ip and firmware stopped after that:

# Serial Log
```
I (50241) bm1366Module: RX Job ID: 28
I (50241) asic_result: Nonce difficulty 821.47 of 1000.
I (50471) bm1366Module: RX Job ID: 28
I (50481) asic_result: Nonce difficulty 399.29 of 1000.
I (55331) bm1366Module: RX Job ID: 40
I (55331) asic_result: Nonce difficulty 368.11 of 1000.
I (55811) bm1366Module: RX Job ID: 40
I (55811) asic_result: Nonce difficulty 324.15 of 1000.
E (58941) stratum_api: recv
I (58941) stratum_api: Restarting System because of Error: recv
I (59951) wifi:state: run -> init (0x0)
I (59951) wifi:pm stop, total sleep time: 44789814 us / 58347512 us

I (59951) wifi:<ba-del>idx:0, tid:0
I (59951) wifi:<ba-del>idx:1, tid:4
I (59951) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (60001) wifi:flush txq
I (60001) wifi:stop sw txq
I (60001) wifi:lmac stop hw txq
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b88
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (31) boot: ESP-IDF v5.1 2nd stage bootloader
I (31) boot: compile time Mar 18 2024 01:01:26
I (31) boot: Multicore bootloader
I (34) boot: chip revision: v0.2
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 16MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00400000
I (91) boot:  3 www              Unknown data     01 82 00410000 00300000
I (99) boot:  4 ota_0            OTA app          00 10 00710000 00400000
I (106) boot:  5 ota_1            OTA app          00 11 00b10000 00400000
I (114) boot:  6 otadata          OTA data         01 00 00f10000 00002000
I (121) boot:  7 coredump         Unknown data     01 03 00f12000 00010000
I (129) boot: End of partition table
I (133) esp_image: segment 0: paddr=00710020 vaddr=3c0b0020 size=2d16ch (184684) map
I (175) esp_image: segment 1: paddr=0073d194 vaddr=3fc9a500 size=02e84h ( 11908) load
I (177) esp_image: segment 2: paddr=00740020 vaddr=42000020 size=aceach (708268) map
I (307) esp_image: segment 3: paddr=007eced4 vaddr=3fc9d384 size=02194h (  8596) load
I (310) esp_image: segment 4: paddr=007ef070 vaddr=40374000 size=1649ch ( 91292) load
I (343) boot: Loaded app from partition at offset 0x710000
I (343) boot: Disabling RNG early entropy source...
I (354) cpu_start: Multicore app
I (364) cpu_start: Pro cpu start user code
I (364) cpu_start: cpu freq: 160000000 Hz
I (364) app_init: Application information:
I (367) app_init: Project name:     esp-miner
I (372) app_init: App version:      v2.1.8
I (377) app_init: Compile time:     Jun  8 2024 10:17:32
I (383) app_init: ELF file SHA256:  92453246f...
I (388) app_init: ESP-IDF:          v5.3-beta1
I (393) efuse_init: Min chip rev:     v0.0
[... 205 lines cut ...]
I (175991) stratum_api: Restarting System because of Error: recv
I (176991) wifi:state: run -> init (0x0)
I (176991) wifi:pm stop, total sleep time: 134193861 us / 174514975 us

I (176991) wifi:<ba-del>idx:0, tid:0
I (176991) wifi:<ba-del>idx:1, tid:2
I (176991) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1, snd_ch_cfg:0x0
I (177041) wifi:flush txq
I (177041) wifi:stop sw txq
I (177041) wifi:lmac stop hw txq
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b88
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (31) boot: ESP-IDF v5.1 2nd stage bootloader
I (31) boot: compile time Mar 18 2024 01:01:26
I (31) boot: Multicore bootloader
```

skot, 2024-06-09:
How long did you wait before rebooting?

pixeldoc2000, 2024-06-09:
@skot the first time it sit stuck like 4h+. Got the issue again tonight, Uptime was 7h (i belive it was stuck like 6h).

skot, 2024-06-17:
this is still happening in 2.1.8. need to merge in @bitMaker-hub fix from #185

WantClue, 2024-12-01:
Fixed in #320

closed 2024-12-01 by WantClue