bitaxeorg/ESP-Miner #193: Nicehash sends separate mining.set_extranonce messages sometimes... maybe handle them?

skot, 2024-05-31:
I haven't seen this before on any other pool. Maybe we should consider handling this unsolicited extranonce change?

```
I (9556) stratum_task: Get IP for URL: sha256asicboost.auto.nicehash.com
I (9586) stratum_task: Connecting to: stratum+tcp://sha256asicboost.auto.nicehash.com:9200 (130.211.20.161)
I (9586) stratum_task: Socket created, connecting to 130.211.20.161:9200
I (9586) ASIC_task: ASIC Ready!
I (9596) main_task: Returned from app_main()
I (9606) stratum_api: tx: {"id": 1, "method": "mining.subscribe", "params": ["bitaxe/BM1368"]}
I (9606) stratum_api: tx: {"id": 2, "method": "mining.configure", "params": [["version-rolling"], {"version-rolling.mask": "ffffffff"}]}
I (9626) stratum_api: tx: {"id": 3, "method": "mining.suggest_difficulty", "params": [1000]}
I (9636) stratum_api: tx: {"id": 4, "method": "mining.authorize", "params": ["<redacted>", "x"]}
I (9696) stratum_task: rx: {"id":1,"error":null,"result":[[["mining.set_difficulty","02187bbb75009a23a3"],["mining.notify","02187bbb75009a23a3"]],"02187bbb75009a23a3",3]}
I (9706) stratum_api: extranonce_str: 02187bbb75009a23a3
I (9706) stratum_api: extranonce_2_len: 3
I (9726) stratum_task: rx: {"id":4,"result":true,"error":null}
I (9726) stratum_task: setup message accepted
I (9726) stratum_task: rx: {"id":null,"method":"mining.set_difficulty","params":[262144.0]}
I (9736) stratum_task: Set stratum difficulty: 262144
I (9746) stratum_task: rx: {"id":null,"method":"mining.set_version_mask","params":["1fffe000"]}
I (9756) stratum_task: Set version mask: 1fffe000

...

I (82806) stratum_task: rx: {"id":null,"method":"mining.set_extranonce","params":["3263b370affeb0d550",3]}
I (82806) stratum_api: unhandled method in stratum message: {"id":null,"method":"mining.set_extranonce","params":["3263b370affeb0d550",3]}
I (82816) stratum_task: Set version mask: 1fffe000
I (82826) stratum_task: rx: {"id":null,"method":"mining.set_difficulty","params":[262144.0]}
```

seangrant82, 2024-06-11:
does this prevent hashing on nicehash? I haven't been able to get it to work

seangrant82, 2024-06-12:
it works on bitaxe but not nerdaxe. will have to submit issue there:

₿ (21067) STRATUM: Socket connection closed
₿ (21067) stratum_task: Failed to receive JSON-RPC line, reconnecting...
₿ (26077) stratum_task: Shutting down socket and restarting...
₿ (26077) stratum_task: Socket created, connecting to 130.211.20.161:9200
₿ (26107) stratum_api: tx: {"id": 9, "method": "mining.subscribe", "params": ["NerdAxe/BM1366"]}
₿ (26217) stratum_api: Received result {"id":9,"error":null,"result":[[["mining.set_difficulty","1111f6c5ce2b7dcdfb"],["mining.notify","1111f6c5ce2b7dcdfb"]],"1111f6c5ce2b7dcdfb",3]}
₿ (26227) stratum_api: tx: {"id": 10, "method": "mining.configure", "params": [["version-rolling"], {"version-rolling.mask": "ffffffff"}]}
₿ (32317) stratum_api: Received result {"id":0,"method":"client.reconnect","params":[]}
₿ (32317) stratum_api: tx: {"id": 11, "method": "mining.suggest_difficulty", "params": [1000]}
₿ (32337) stratum_api: tx: {"id": 12, "method": "mining.authorize", "params": ["wallet.nerdaxe", "fake_password"]}
₿ (32347) STRATUM: Socket connection closed
₿ (32347) stratum_task: Failed to receive JSON-RPC line, reconnecting...

seangrant82, 2024-06-12:
Doesn't seem to accept shares though 
[image]