bitaxeorg/ESP-Miner #199: esp-miner restarts after some hours when AxeOS is left connected

skot, 2024-06-04:
#182 fixed a number of restart errors, but this one still remains. In some cases the restart happens as soon as a websocket connection is made.

benjamin-wilson, 2024-06-04:
I've tried for 30min to get an error out of the websocket or http server but was unable. If someone has logs that'd be helpful.

Shen, 2024-06-05:
I got one:

```
I (1652169) stratum_task: rx: {"result":true,"error":null,"id":441}
I (1652169) stratum_task: message result accepted
I (1652849) bm1368Module: Job ID: 9E
I (1652849) bm1368Module: RX Job ID: 48
I (1652849) asic_result: Nonce difficulty 463.20 of 479.
I (1653319) stratum_task: rx: {"params":["648516380010a383","26c604aac06ee1966425490d8f12d4df01b295e4000142e90000000000000000","01000000010000000000000000000000000000000000000000000000000000000000000000ffffffff350305eb0c0004540e60660442e8f3130c","0a636b706f6f6c112f736f6c6f2e636b706f6f6c2e6f72672fffffffff035262a813000000002200201bc038ee5f8759ad75eb703c00d860de2ac75fdb6c2852424845ee1998927da4a3b366000000000016001451ed61d2f6aa260cc72cdf743e4e436a82c010270000000000000000266a24aa21a9ed60c4d513c43a6c611793399ecde11fc5cc5b0b28c049239d7d6b50e203fbb1bf00000000",["6e2cb12ab468ff1e13ccfd55c290d8a0d5fe7f56a48f0d744a763600de3d3cd6","16af7a609116b0526eb4cba78dc433dbddb0ec479beb8f0b171f53ab9ac429f4","82748581976bdf6e97a6e057b124ec4cf1521888e36aad3dc29ab303e28a5bc1","022521e2e2a54400e6a71ef17560bc2d4a2f6822a6bf17bd39c5a927ac19ec32","5ce64ad7cb44b856f18656ac2e8c21861b9d77fc3274caf6f7aa21e4f275b3b2","f35addc74fb38c79669f154fc3ffaaa4d7ba444564d500d7e9b7a07c583687e4","72705aff85ad8ae889e1181e9878c0ebe4aba8285a9c2c5b3a2f25b124c3a1e1","6fe0f7a36bd6cc5430a3e4a503c06e9d19c22909957ae4b0865231329e998802","52c08aaea04d7cac4d14feda2fa7d1b2065524176372624221f176684d474d03","d6bcd93c59654ff2211b28a55d6094e8ab01bc026bc651cc9ceba5d679026aee","9a5ea3d403a4fb19083df7006a0fd62d36f9a5708afb7ab0137fc239fd248c59","9b27f4e3ec17fea297207ddea63ab5b0b9d3f2155324155c96044042b84cdd2d","35c9d7a14fdefcd2e63b4bc72bfcab85bcaa9d6e4414921579867164aaaa7d25"],"20000000","170355f0","66600e54",false],"id":null,"method":"mining.notify"}
I (1653349) bm1368Module: Job ID: F7
I (1653439) bm1368Module: RX Job ID: 78
I (1653449) asic_result: Nonce difficulty 1962.69 of 479.
I (1653459) create_jobs_task: New Work Dequeued 648516380010a383
W (1653459) httpd_txrx: httpd_sock_err: error in send : 104
W (1653459) httpd_txrx: httpd_sock_err: error in send : 128
W (1653469) httpd_txrx: httpd_sock_err: error in send : 128
W (1653479) httpd_txrx: httpd_sock_err: error in send : 128
W (1653479) httpd_txrx: httpd_sock_err: error in send : 128
W (1653479) httpd_txrx: httpd_sock_err: error in send : 128
W (1653499) httpd_txrx: httpd_sock_err: error in send : 128
W (1653499) httpd_txrx: httpd_sock_err: error in send : 128
W (1653509) httpd_txrx: httpd_sock_err: error in send : 128
W (1653509) httpd_txrx: httpd_sock_err: error in send : 128
W (1653519) httpd_txrx: httpd_sock_err: error in send : 128
W (1653519) httpd_txrx: httpd_sock_err: error in send : 128
W (1653529) httpd_txrx: httpd_sock_err: error in send : 128
W (1653529) httpd_txrx: httpd_sock_err: error in send : 128
W (1653549) httpd_txrx: httpd_sock_err: error in send : 128
W (1653549) httpd_txrx: httpd_sock_err: error in send : 128
W (1653559) httpd_txrx: httpd_sock_err: error in send : 128
W (1653559) httpd_txrx: httpd_sock_err: error in send : 128
W (1653569) httpd_txrx: httpd_sock_err: error in send : 128
W (1653579) httpd_txrx: httpd_sock_err: error in send : 128
W (1653579) httpd_txrx: httpd_sock_err: error in send : 128
W (1653579) httpd_txrx: httpd_sock_err: error in send : 128
W (1653599) httpd_txrx: httpd_sock_err: error in send : 128
W (1653599) httpd_txrx: httpd_sock_err: error in send : 128
W (1653609) httpd_txrx: httpd_sock_err: error in send : 128
W (1653609) httpd_txrx: httpd_sock_err: error in send : 128
W (1653619) httpd_txrx: httpd_sock_err: error in send : 128
W (1653629) httpd_txrx: httpd_sock_err: error in send : 128
W (1653629) httpd_txrx: httpd_sock_err: error in send : 128
W (1653639) httpd_txrx: httpd_sock_err: error in send : 128
W (1653639) httpd_txrx: httpd_sock_err: error in send : 128
W (1653649) httpd_txrx: httpd_sock_err: error in send : 128

***ERROR*** A stack overflow in task asic result has been detected.

Backtrace: 0x40375c42:0x3fccc1b0 0x4037d819:0x3fccc1d0 0x4037e7d2:0x3fccc1f0 0x4037fca7:0x3fccc270 0x4037e898:0x3fccc290 0x4037e88e:0x00000000 |<-CORRUPTED

ELF file SHA256: 4ae5e3ac7675b79f

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2a (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375b7c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x16e0
load:0x403c9700,len:0x4
load:0x403c9704,len:0xc00
load:0x403cc700,len:0x2eb0
entry 0x403c9908
I (26) boot: ESP-IDF v5.1 2nd stage bootloader
I (26) boot: compile time Mar  3 2024 19:55:36
I (26) boot: Multicore bootloader
I (29) boot: chip revision: v0.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (38) boot.esp32s3: SPI Mode       : DIO
[... 156 lines cut ...]
I (21529) bm1368Module: Job ID: F1
I (21529) bm1368Module: RX Job ID: 78
I (21529) asic_result: Nonce difficulty 546.91 of 10000.
I (21639) bm1368Module: Job ID: 0A
I (21639) bm1368Module: RX Job ID: 00
I (21639) asic_result: Nonce difficulty 445.61 of 10000.
I (22819) bm1368Module: Job ID: 26
I (22819) bm1368Module: RX Job ID: 10
I (22819) asic_result: Nonce difficulty 422.61 of 10000.
I (23479) bm1368Module: Job ID: 83
I (23479) bm1368Module: RX Job ID: 40
I (23479) asic_result: Nonce difficulty 1166.99 of 10000.
I (23489) bm1368Module: Job ID: BF
I (23489) bm1368Module: RX Job ID: 58
I (23489) asic_result: Nonce difficulty 1767.21 of 10000.
I (24329) bm1368Module: Job ID: 75
I (24329) bm1368Module: RX Job ID: 38
I (24329) asic_result: Nonce difficulty 665.04 of 10000.
[... cut ...] 79bc1b35471f333cbcc","9b1e33dcd34c26bd7777998be50116f462c8803a2b7b97d0a5950a29a7fbb058","380c4f72bcab87b8b56716d2728b12fc0099992966efedf60d47c4d9f8c80955","e9bde485a6d31daef033f48f3c02c94411df01b96a48482bdf9574bff11644b7","6e0c6daef852de3617d38238e0ce95f9da1448a86c201031961c3486f2d0cf74","8814515d93ce0cb3d50957b739e109a89d1f79a10e37d219c60d1d52f14b3af5"],"20000000","170355f0","66600e72",false],"id":null,"method":"mining.notify"}
I (29639) create_jobs_task: New Work Dequeued 648516380010a384

```

Shen, 2024-07-11:
Do you have a potential solution for this problem in sight?

solobito, 2024-10-10:
This has already been fixed in version 2.2.2. Maybe due to these changes: "_Fix websocket logs causing device to crash #277 by @eandersson in #318_"

closed 2024-10-10 by skot