bitaxeorg/ESP-Miner #207: New Work Dequeued

Rafal4265, 2024-06-07:
- Bitaxe HW version: Supra 401
 - Bitaxe HW vendor: nerdminer store
 - ESP-Miner FW version: 2.1.7
- Pool URL: solo.ckpool.org

Hi. After purchase, the equipment worked very well but after upgrading to the latest software version no hash, I do not remember what was the previous version, logs:

₿ (951414) stratum_task: rx: {"params":["648516380010bb7f","92af392b20c368d0c842e2f37b9df88bc178a22300025b220000000000000000","01000000010000000000000000000000000000000000000000000000000000000000000000ffffffff35030cec0c000461bf626604178465090c","0a636b706f6f6c112f736f6c6f2e636b706f6f6c2e6f72672fffffffff034f074b14000000001600148bd480a8577a64362975a61edbfd6e10dcf5935f5f056a000000000016001451ed61d2f6aa260cc72cdf743e4e436a82c010270000000000000000266a24aa21a9edb16c4eba813f36f53322dafc5a576e62bff1ff2c3ca3175f2138e362298054d100000000",["3275137b8bb74c548efec2ba0918d754632e704d1d31390a37d6e4ebe2ff4208","e8f39f474cb38d3f7467acd6cd77409ecca9f618f539b09ca87c2939530120da","4de21e7e3b764ee7e8b4d886023e35d8c7fe62ed27b1a3f49194edc144719ab3","5cb08347a6ff6b33f0bc91b6fe8e6049316498e4ed6b2c3e85db7da7a24862f9","92cf6a6d9fb211f5815891342f61b31a7dfba25e31d1f876e9586da4440c7928","48e7d2bb6b4053ef5c1f1f2a75ec23309d02be44d0408ba6a8277708a89fe8f9","1fff6406e7e49bb337521b0e88b98e8d45b934b548f7329075a6db1753b94bae","02e81c7291a86c29726ec014e80f59f3177aaf4581db2e5d2ae1457d637438fe","114b37d7629f00ff99dd4ff24272bfad47326b0256013c2fbde7e3f08a56b0a0","73b4bddb5fea7f50937a2c5812ee2165a3053ed60aaa5ac50c0261b1408d2c16","391a0c548763dfe8243f5680e7d3b74b1cc482e6eac2f5d3ef8698f4ea9432bb","aad95fe06c29e494266405b8d82239f206f516cb5bb8486fe37b554fc53117d8"],"20000000","17035cb8","6662bf61",false],"id":null,"method":"mining.notify"}
₿ (951654) create_jobs_task: New Work Dequeued 648516380010bb7f

Rafal4265, 2024-06-07:
> ₿ (951414) stratum_task: rx: [...]
₿ (951654) create_jobs_task: New Work Dequeued 648516380010bb7f

weixiongmei, 2024-06-08:
Do you know what chip you have? BM1366AL or BM1366AG?

skot, 2024-06-08:
Please contact the seller or use the OSMU help channel for support.

closed 2024-06-08 by skot

Rafal4265, 2024-06-08:
Model:	BM1368