bitaxeorg/ESP-Miner #212: Batch of rejections after stratum difficulty adjustment due to 'Above target' and reboot

mutatrum, 2024-06-10:
Running a BitAxe Supra (400), firmware 2.1.8 on solo.ckpool.org.

After startup, initial stratum difficulty is 10000. It takes a fair bit of time to find a handful of shares. After this, the difficulty is adjusted downwards, but this results in a batch of rejected shares, with an error message of 'Above target'. This cycle repeats.

After this, it starts accepting shares again, way faster than before and then it reboots. The end of the logfile is the reboot. The log is captured with websocat, but it's cut short due to an UTF-8 failure (separate issue maybe?). I'll see if I can also capture a log from serial.

[bitaxe2.log](https://github.com/user-attachments/files/15775643/bitaxe2.log)

[image]

skot, 2024-06-10:
rejecting work after _they_ lowered the difficulty really seems like a ck-pool bug. The developer has been notified, but we haven't gotten a response, afaik. 

If you can get the log (prolly with USB) from where the restart happens, that would be helpful. I'm not sure what that's about.
```
I (381191) stratum_task: rx: {"params":[398],"id":null,"method":"mining.set_difficulty"}
I (381191) stratum_task: Set stratum difficulty: 398
I (381201) stratum_task: rx: {"result":true,"error":null,"id":9}
I (381201) stratum_task: message result accepted
I (381291) bm1368Module: Job ID: A3
I (381291) bm1368Module: RX Job ID: 50
I (381291) asic_result: Nonce difficulty 757.35 of 10000.
I (383461) bm1368Module: Job ID: 57
I (383471) bm1368Module: RX Job ID: 28
I (383471) asic_result: Nonce difficulty 506.16 of 10000.
I (393681) stratum_task: rx: {"params":["648516380010e299","90ab7ca18a009bb20b05e8dfa5b5688bf717b076000070410000000000000000","01000000010000000000000000000000000000000000000000000000000000000000000000ffffffff350301ee0c0004ba18676604a1ab11160c","0a636b706f6f6c112f736f6c6f2e636b706f6f6c2e6f72672fffffffff037b11c9140000000016001480bded37e3f86a1a546e099b15e2b02855d3843cdd976c000000000016001451ed61d2f6aa260cc72cdf743e4e436a82c010270000000000000000266a24aa21a9ed509253c2c42d27c26dcbb540cf71f9ca7ff4e1777ed513ccefda80dec886885100000000",["8c299541de94a07e5b27f7c50d4835bf77cf75789042c6ec4c5ae7345e821d31","173abf20a4f11aaa1850679b1117e0b3497bf9e76030cee150fbb8b2f2f1f019","01fe12b0401efa90494beaaece8b387bc89f697dee8104cdafe3c4725bed7bf6","07b4a76b49c36165c9aa3510cd9629a00c6ffbad075b4beeee92b0b12760f763","29a624e985273cb00fea48675be395f5daa0a48ab7b324056a8fbd6e97bde1a6","46979615d1a33db96d76fcbf1d2bd26d660dc940b1b4df4fbb9206c7f4a1de5b","625ae540510a9007be34731c2cc6f4cd98a4cd2b25fab6bf85961b0447ce8964","b367ee622207571ddd708f99bccd06f674f5116618864650c3aa5015a356a67e","a52f313ca0ef691b33be42cf0ce301970f0d062bc05bf6679f3baacc890fea00","e7462bc6a30bec58be50dfdf5c2a973baaed26096ca38c3935f641a7bc10ecf4","f737fc1c2a3ba149adea4da799f0f7b8c37d3ac9da665f7adf340fa619305bcd","fde435cdba35eb4d6017f89d19886200bcff14b4c52ebe26e445abdc0700ccdc"],"20000000","17035cb8","666718ba",false],"id":null,"method":"mining.notify"}
I (393811) create_jobs_task: New Work Dequeued 648516380010e299
I (393831) bm1368Module: Job ID: 92
I (393831) bm1368Module: RX Job ID: 48
I (393831) asic_result: Nonce difficulty 1872.60 of 10000.
I (393941) ASIC_task: New pool difficulty 398
I (397101) bm1368Module: Job ID: EB
I (397101) bm1368Module: RX Job ID: 70
I (397101) asic_result: Nonce difficulty 581.89 of 398.
I (397101) stratum_api: tx: {"id": 10, "method": "mining.submit", "params": ["bc1qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq.bitaxe", "648516380010e299", "3b01000000000000", "666718ba", "0c7e4366", "00016000"]}
I (397261) stratum_task: rx: {"reject-reason":"Above target","result":false,"error":null,"id":10}
E (397271) stratum_task: message result rejected
I (397581) bm1368Module: Job ID: 14
I (397591) bm1368Module: RX Job ID: 08
I (397591) asic_result: Nonce difficulty 291.87 of 398.
I (397931) bm1368Module: Job ID: A5
I (397941) bm1368Module: RX Job ID: 50
I (397941) asic_result: Nonce difficulty 611.18 of 398.
I (397941) stratum_api: tx: {"id": 11, "method": "mining.submit", "params": ["bc1qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq.bitaxe", "648516380010e299", "8f01000000000000", "666718ba", "f3f92a56", "0000a000"]}
I (398081) stratum_task: rx: {"reject-reason":"Above target","result":false,"error":null,"id":11}
```

mutatrum, 2024-06-10:
Got the reboot from serial:

```I (1284861) stratum_task: rx: {"result":true,"error":null,"id":248}
I (1284861) stratum_task: message result accepted
I (1286591) bm1368Module: Job ID: 03
I (1286591) bm1368Module: RX Job ID: 00
I (1286601) asic_result: Nonce difficulty 443.31 of 533.
I (1287521) stratum_task: rx: {"params":["648516380010e2e8","eab23622245775c0ffaf97ff25bf4eda0db7e7790001de030000000000000000","01000000010000000000000000000000000000000000000000000000000000000000000000ffffffff350309ee0c0004f320676604eb463f0c0c","0a636b706f6f6c112f736f6c6f2e636b706f6f6c2e6f72672fffffffff033da6f1130000000016001480bded37e3f86a1a546e099b15e2b02855d3843c693268000000000016001451ed61d2f6aa260cc72cdf743e4e436a82c010270000000000000000266a24aa21a9ed2b913d904d4a0f0db9f5ed947e75255edcfa48ea4486d8ef10316f9ce75e902400000000",["f6c8995a2a8b389145a48d225cc370d2cb6b44391ea2cb606e1491078e487641","909bc877ac6ebcdd58857c2aff8ee7ca4ba001013c19033e49f2a4a44b42a2f8","a4b164da13318d927be9efde01397002ecb62d5ed3cca646cde87407383bd32a","321d298ada07ffad525c92e44160f915976ed980cb3e6fc7d04ea74d2ae6f7ac","516ff2ffed3a93ca1cba45a05b32526ffc0facf90b9d51638a881e7b2a8a60bc","c3fe2eef74827054e2dfb42260ae3e67d5a2de2cf3139c4c721b70a0ad8d5858","79019f88859bce2820f2fb2dc0980bc3b5817b4a9f71f4670d76ff1d1d6a6e38","2d7dbceb096d173beecd97de2c91d6b4e1be9c450062c9fb706be7c3c1484df3","cda86429c6fac89da360646bbfc39d4e61fb3162e316487033671ed6ce722508","15f7d3772df3ec9a0b3b6057cedde384150146e6ce2e281e0955a82385a55c73","3f3358571a69fe8dc43f47733af1a00e1e801aff1d5de13761ebf527a584564e","cd55d0c2ea3432c604ed5b685912847014f57319cf577bcc284f1f9a1844497d","be1b5e728cb5328c7095f300c4f79ba9ce68fc929be7ac531328ec0d2c4b235e"],"20000000","17035cb8","666720f3",false],"id":null,"method":"mining.notify"}
I (1287571) bm1368Module: Job ID: 6E
I (1287651) bm1368Module: RX Job ID: 30
I (1287651) asic_result: Nonce difficulty 3448.98 of 533.
I (1287661) stratum_api: tx: {"id": 249, "method": "mining.submit", "params": ["bc1qqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqqq.bitaxe", "648516380010e2e7", "a30b000000000000", "666720d5", "00c52e3e", "0001c000"]}

W (1287681) httpd_txrx: httpd_sock_err: error in send : 128
I (1287681) create_jobs_task: New Work Dequeued 648516380010e2e8
W (1287681) httpd_txrx: httpd_sock_err: error in send : 128
W (1287691) httpd_txrx: httpd_sock_err: error in send : 128
W (1287701) httpd_txrx: httpd_sock_err: error in send : 128
W (1287711) httpd_txrx: httpd_sock_err: error in send : 128
W (1287711) httpd_txrx: httpd_sock_err: error in send : 128
W (1287721) httpd_txrx: httpd_sock_err: error in send : 128
W (1287721) httpd_txrx: httpd_sock_err: error in send : 128
W (1287731) httpd_txrx: httpd_sock_err: error in send : 128
W (1287731) httpd_txrx: httpd_sock_err: error in send : 128
W (1287741) httpd_txrx: httpd_sock_err: error in send : 128
W (1287751) httpd_txrx: httpd_sock_err: error in send : 128
W (1287761) httpd_txrx: httpd_sock_err: error in send : 128
W (1287761) httpd_txrx: httpd_sock_err: error in send : 128
W (1287771) httpd_txrx: httpd_sock_err: error in send : 128
W (1287771) httpd_txrx: httpd_sock_err: error in send : 128
W (1287671) httpd_txrx: httpd_sock_err: error in recv : 104
W (1287781) httpd_txrx: httpd_sock_err: error in send : 128
W (1287781) httpd_txrx: httpd_sock_err: error in send : 128
W (1287801) httpd_txrx: httpd_sock_err: error in send : 128
W (1287811) httpd_txrx: httpd_sock_err: error in send : 128
W (1287811) httpd_txrx: httpd_sock_err: error in send : 128
W (1287821) httpd_txrx: httpd_sock_err: error in send : 128
W (1287821) httpd_txrx: httpd_sock_err: error in send : 128
W (1287801) httpd_txrx: httpd_sock_err: error in send : 128
W (1287831) httpd_txrx: httpd_sock_err: error in send : 128
W (1287831) httpd_txrx: httpd_sock_err: error in send : 128
W (1287851) httpd_txrx: httpd_sock_err: error in send : 128

***ERROR*** A stack overflow in task asic result has been detected.

Backtrace: 0x40375c4e:0x3fccbf70 0x4037d841:0x3fccbf90 0x4037e7fa:0x3fccbfb0 0x4037fcbf:0x3fccc030 0x4037e8c0:0x3fccc050 0x4037e8b6:0x00000000 |<-CORRUPTED

ELF file SHA256: 92453246f

Rebooting...
```

It might be more connected to the websocket connection? It ran for quite a while, and rebooted a few minutes after I also opened the websocket logs. I have more logs, but what happened before that didn't look relevant.

mutatrum, 2024-06-10:
This is from websocat, to line it up:
```
I (1284671) bm1368Module: Job ID: 02
I (1284671) bm1368Module: RX Job ID: 00
I (1284681) asic_result: Nonce difficulty 593.43 of 533.
I (1284681) stratum_api: tx: {"id": 248, "method": "mining.submit", "params": ["bc1qsz776dlrlp4p54rwpxd3tc4s9p2a8ppuceyvdc.bitaxe", "648516380010e2e7", "810a000000000000", "666720d5", "e4fd1f0c", "00004000"]}

I (1284861) stratum_task: rx: {"result":true,"error":null,"id":248}
I (1284861) stratum_task: message result accepted
I (1286591) bm1368Module: Job ID: 03
I (1286591) bm1368Module: RX Job ID: 00
I (1286601) asic_result: Nonce difficulty 443.31 of 533.
I (1287571) bm1368Module: Job ID: 6E
I (1287651) bm1368Module: RX Job ID: 30
websocat: WebSocketError: UTF-8 failure
websocat: error running
```

mutatrum, 2024-06-10:
Addition: I also have a script that polls `/api/system/info` once a minute. Seems to coincide with #199.

I'm running it again now, without the websocket but with the USB logging capture.

skot, 2024-06-10:
okay, it seems like the restart is a separate issue (#199) then

skot, 2024-06-10:
I'm pretty sure I have seen this same issue last year where ckpool lowers the difficulty, and then doesn't start accepting shares with that new lower difficulty until the _next_ mining.notify

kakawlala, 2024-06-10:
pool.vkbit.com also has a default difficulty of 10,000.  But currently the hash calculation appears to be local instead of waiting for the pool to feed back.  This is helpful.  I know it's working and the difficulty eases off after a while.  Although I get some rejected parts.  But when connected for a long time, the rejected part is less than 1%

This is the data of supra401 running on pool.vkbit.com for 43 hours（2.1.8.bin）
[image]

skot, 2024-06-10:
vkbit is a ckpool clone, so I suspect it will have this same issue.

mutatrum, 2024-06-11:
I have it running now for ~18 hours, it looks like ckpool only requests a new difficulty once, a few minutes after start of the BitAxe. The only rejected shares (12 in my case) were from that initial ramp down, after that no rejected shares. So for ckpool the impact is minor indeed.

skot, 2024-06-11:
removed the bug tag. afaik this isn't something we can fix on the esp-miner side.

closed 2025-01-27 by WantClue