bitaxeorg/ESP-Miner #214: Inaccessible AxeOS Dashboard

crimsonleaf363, 2024-06-10:
**Describe the bug**
The AxeOS dashboard is currently inaccessible from the local IP address displayed on the Bitaxe screen. The connection attempt fails.

**To Reproduce**
Steps to reproduce the behavior:
1. Ensure the Bitaxe device is powered on and functioning correctly.
2. Observe the local IP address displayed on the Bitaxe screen.
3. Attempt to access the AxeOS dashboard using the local IP address.

**Expected behavior**
The user should be able to access the AxeOS dashboard from the local IP address displayed on the Bitaxe screen.

**Hardware (please complete the following information):**
 - Bitaxe HW version: Ultra 204
 - ESP-Miner FW version: 2.1.6
 - Hash Frequency: N/A (I don't know how to determine the hash frequency.)
 - Voltage: N/A (I don't know how to determine the voltage.)
 - Pool URL, Port, User: Default

19201003080114, 2024-06-10:
Does your router settings allow local network traffic between devices?
If you connect Bitaxe to the “Guest” network, you won’t be able to access its IP. 
I can access my Bitaxe local IP without issues.

crimsonleaf363, 2024-06-10:
> Does your router settings allow local network traffic between devices? If you connect Bitaxe to the “Guest” network, you won’t be able to access its IP. I can access my Bitaxe local IP without issues.

I'm able to access other local IPs between different devices.

crimsonleaf363, 2024-06-11:
> you could try and turn off your router then power it on. Once it cant find your wifi, it will broadcast its own AP. Connect your comp to that and see if you can see it. If so, update FW to 2.1.8 and ✅✅ pool settings. I messed up my pool settings and it wouldn't let me access it until i fixed it.

This didn't fix my issue.

crimsonleaf363, 2024-06-11:
> Does your router settings allow local network traffic between devices? If you connect Bitaxe to the “Guest” network, you won’t be able to access its IP. I can access my Bitaxe local IP without issues.

I updated my firmware to 2.1.8, but that did not fix the issue. After I switched the Bitaxe to a non-guest Wi-Fi, the problem was solved. I'm able to access the AxeOS dashboard without a problem. Thank you.

skot, 2024-06-11:
ok that makes sense. Guest WiFi doesn't usually let clients communicate with each other