bitaxeorg/ESP-Miner #246: Exclude pre releases of updated check

WantClue, 2024-07-02:
With the latest pre-release which has been removed now we had experienced the issue, that certain people updated.
In order to fight this issue of not downloading pre-releases I suggest an change to the github-update.service.ts

The change would be as followed:
`import { HttpClient } from '@angular/common/http';
import { Injectable } from '@angular/core';
import { Observable } from 'rxjs';
import { map } from 'rxjs/operators';

interface GithubRelease {
  tag_name: string;
  prerelease: boolean;
  // Add other properties as needed
}

@Injectable({
  providedIn: 'root'
})
export class GithubUpdateService {

  constructor(
    private httpClient: HttpClient
  ) { }

  public getReleases(): Observable<GithubRelease[]> {
    return this.httpClient.get<GithubRelease[]>(
      'https://api.github.com/repos/skot/esp-miner/releases?per_page=100'
    );
  }

  public getLatestStableRelease(): Observable<GithubRelease | null> {
    return this.getReleases().pipe(
      map(releases => releases.find(release => !release.prerelease) || null)
    );
  }
}`

in settings.components.ts:

`this.latestRelease$ = this.githubUpdateService.getLatestStableRelease();`

skot, 2024-07-02:
This would be amazing. I think pre-releases are a good way to roll out firmware to more experienced users who are willing to beta test new firmware.

WantClue, 2024-08-19:
Done in PR #285

closed 2024-08-19 by WantClue