bitaxeorg/ESP-Miner #251: Please add compatibility with Nicehash

amauryvsosa, 2024-07-04:
Could You please add compatibility to conect to nicehash pool?

skot, 2024-07-04:
what's not working?

amauryvsosa, 2024-07-04:
Hi skot, good morning/afternoon/night
I'm using Bit Axe Supra v401
I'm using the generated pool by nicehash appears connected but in the logs
it shows err no 113 and tries to connect to 0.0.0.0
And using Xmrig with the same generated pool and port in a Android device
nicehash recognice it

I can connect the bitaxe to public pool io and vkbit

skot, 2024-07-04:
ok, 0.0.0.0 is a DNS error. are you sure you entered the stratum address correctly?

amauryvsosa, 2024-07-04:
Hi Skot
To be more precise I show you the configuration

stratum given by nicehash and bitaxe config
[image: configuracion.png]

using port 443 logs
[image: WhatsApp Image 2024-07-04 at 12.59.38 PM.jpeg]

using port 9200 there are no logs
[image: WhatsApp Image 2024-07-04 at 12.59.39 PM.jpeg]

hash rate, shares in both cases Nicehash doesn't recognize the device
and in both cases the device keeps rebooting.

I'm doing something wrong? or maybe it's a bug
I would really appreciate your help

skot, 2024-07-04:
the images didn't post

amauryvsosa, 2024-07-04:
An apology

Stratum generated by Nicehash and configuration <https://ibb.co/KqXJWjD> (
https://ibb.co/KqXJWjD)

Port 443 logs <https://ibb.co/PjfDgqB> (https://ibb.co/PjfDgqB)

Port 9200 Logs <https://ibb.co/g9p14B6> (https://ibb.co/g9p14B6)

skot, 2024-07-04:
can you try and get the logs from `stratum+tcp://sha256asicboost.auto.nicehash.com:9200` 443 is SSL and isn't supported.

skot, 2024-07-04:
for me, nicehash sets the initial difficulty to `262144.0` so it will take a while on a Bitaxe before any shares are submitted.

amauryvsosa, 2024-07-04:
so, do you can connect your miner to nicehash correctly?

Configuration with 9200 port <https://ibb.co/rQm9kDL> (
https://ibb.co/rQm9kDL)

Hashrate <https://ibb.co/hZ4qf1w> (https://ibb.co/hZ4qf1w)

I left it running for about 30 minutes but it keeps restarting
did you make any extra configuration in your router or something?

amauryvsosa, 2024-07-04:
update:
I was already able to connect with Nicehash by updating the firmware from
2.1.3 to 2.1.8 and it recognizes the device, however, shares and rejected
remain at 0

CoanLuciano, 2024-07-05:
@skot check these logs collected today. 

Model:	BM1366
Version:	v2.1.8
Board Version:	0.11

₿ (15574) stratum_api: tx: {"id": 5, "method": "mining.submit", "params": ["<my user, I changed>", "00000000a014704b", "05000000", "6687d3de", "a8a0028c", "0b0a6000"]}
₿ (15794) stratum_task: rx: {"id":5,"result":false,"error":[5,"Invalid extranonce2 size."]}
₿ (15794) stratum_task: message result rejected

seangrant82, 2024-07-17:
extra nonce is known issue (https://github.com/skot/ESP-Miner/issues/193) but like @skot mentioned, the difficulty may mean it will take a while for share to be submitted. How long have you had the bitaxe mining to nicehash and still have not see shares?

WantClue, 2024-08-14:
Works on 2.1.10

closed 2024-08-14 by WantClue