bitaxeorg/ESP-Miner #262: Bitaxe httpd crashes

IO-igiannako, 2024-07-29:
**Describe the bug**
After a couple of days (2-3 max) the Bitaxe http server stops responding to http requests. I can't access the web interface and therefore no further information on Bitaxe condition, change settings, check logs, etc. I bought it a week ago and this is the third time in a row it crashes. Bitaxe reponds to ping but doesn't serve http at all. Unfortunately, only hardware restart solves the issue.

**To Reproduce**
N/A, http interface is unavailable after 48-72 hours of uptime.

**Expected behavior**
Theoretically, I expect the web server to constantly serve content, dashboard, settings, logs etc

**Screenshots & Photos**
[image]
[image]
[image]

**Hardware (please complete the following information):**
 - Bitaxe HW version: Ultra 204
 - Bitaxe HW vendor: Bitronics, Spain
 - ESP-Miner FW version: 2.1.8
 - Hash Frequency: ~450 daily average
 - Voltage: 5 (default)
 - Pool URL, Port, User: ckpool 3333 bc1qhsp5y8evd2d4s0axezqn45sqpenm7vzvnzk4tv.Bitaxe

**Additional context**
Add any other context about the problem here.

19201003080114, 2024-07-29:
I had similar issue. After adding heatsinks to ESP32 the webpage is now always available

IO-igiannako, 2024-07-30:
Easy, I'll try that. Cheers

benjamin-wilson, 2024-07-30:
What type of wireless router do you have?

IO-igiannako, 2024-07-30:
The default one the local internet provider offers. It's the Sercomm SHG3060 or Vodafone Power Station WiFi6, as it's branded. I'm not quite sure of the question to be honest, I find it to be a decent WiFi 6 router, quite stable. I'm on the 2.4 band, statistics are stable as well, the ping with ckpool is not excellent but good enough (120-160ms) and I have only 1 share rejected out of 1000 on average, meaning ~0,1%. 

Let me know if there's something more specific, happy to share with you.

IO-igiannako, 2024-07-31:
I installed a heatsink to ESP32 and now seems to be stable. Thanks @19201003080114

kakawlala, 2024-08-08:
Replacing esp32 still doesn't work.
I am placing metal clips
[image]
[image]
[image]

skot, 2024-08-10:
there are some memory leak fixes in https://github.com/skot/ESP-Miner/tree/219-leak_hunting that might fix this. can you give it a try?

[esp-miner.bin.zip](https://github.com/user-attachments/files/16568480/esp-miner.bin.zip)

closed 2024-08-14 by WantClue