bitaxeorg/ESP-Miner #272: Create a general status task

skot, 2024-08-07:
I'd like to create a general mining status indicator for esp-miner. Currently it's too hard to tell if your Bitaxe is actively mining. This would probably be a separate rtos task that would always show the current status on the Bitaxe display and on the AxeOS dashboard. Maybe this takes the form of the watchdog task?

things to monitor:
- have we received a nonce from the ASIC in the expected interval?
- have we received work from the pool?
- is the network socket still open?
- is wifi connected?
- has the bitaxe overheated?
- what else?

WantClue, 2024-08-07:
something like this status: O.K. or status: mining that is been used by the standard bitmain firmware I assume. 

Sure doable. 

I think for a general status O.K. to pass we need all the above to be true execpt for the overheat_mode.

change:

- have we received work from the pool ? --> when was the last time we have received work < threshold

skot, 2024-08-07:
maybe even a little ✅ or ❌ icon for the display in the corner

WantClue, 2024-08-07:
Or we bring back the status LED like on a control board