bitaxeorg/ESP-Miner #277: Websocket (logs) can crash the device

eandersson, 2024-08-08:
If you open too many connections to get logs you'll eventually crash the device and it won't recover on its own. It's pretty easy to reproduce by just having something like a python script open a few ws connections in rapid succession. This seems to be a pretty known issue with ESP32, but wasn't able to find a solution for ws. I tried adding `Connection: close` to the header and set things like `lru_purge_enable` without success.

```
W (1776029) httpd_txrx: httpd_sock_err: error in send : 104
...
W (1776089) httpd_txrx: httpd_sock_err: error in send : 128

***ERROR*** A stack overflow in task httpd has been detected.

Backtrace: 0x40375ada:0x3fcbbae0 0x4037de6d:0x3fcbbb00 0x4037ed76:0x3fcbbb20 0x4038022f:0x3fcbbba0 0x4037ee84:0x3fcbbbc0 0x4037ee7a:0x00000000 |<-CORRUPTED
```

skot, 2024-08-08:
Ah, good catch. Do you know if we can set a connection limit?

eandersson, 2024-08-09:
The issue seems to be with any type of broken connection and not necessarily just having a bunch of connections. This is the code I used to reproduce the issue. I basically just started a few copies in parallel and then killed them using something like `pkill`.

```
import time
from websockets.sync import client
import logging

MY_IP = '192.168.0.100'
LOGFILE = 'my_log_name.log'

DEFAULT_SLEEP_INTERVAL = 1
LOG = logging.getLogger(__name__)
logging.basicConfig(
    filename=LOGFILE,
    encoding='utf-8',
    level=logging.INFO,
    format='%(asctime)s %(message)s',
    datefmt='%m/%d/%Y %I:%M:%S %p'
)

def main():
    while True:
        try:
            with client.connect(f"ws://{MY_IP}/api/ws") as websocket:
                while True:
                    message = websocket.recv(timeout=60)
                    LOG.info(message)
        except Exception as why:
            LOG.warning('Lost connection: %s', why)
            time.sleep(DEFAULT_SLEEP_INTERVAL)

if __name__ == '__main__':
    main()

```

eandersson, 2024-08-09:
The scary part here is that it isn't able to recover on it's own. Pretty sure I managed to trigger this at least once using the browser and with poor WiFi reception.

This is the watchdog crash loop that I get stuck in.
```
Backtrace: 0x40378CD6:0x3FC9D8B0 0x403776DD:0x3FC9D8D0 0x420109EE:0x3FCC6DB0 0x4037E421:0x3FCC6E20

E (41541) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (41541) task_wdt:  - IDLE1 (CPU 1)
E (41541) task_wdt: Tasks currently running:
E (41541) task_wdt: CPU 0: IDLE0
E (41541) task_wdt: CPU 1: stratum admin
E (41541) task_wdt: Print CPU 1 backtrace

Backtrace: 0x40378CD6:0x3FC9D8B0 0x403776DD:0x3FC9D8D0 0x420109EE:0x3FCC6DB0 0x4037E421:0x3FCC6E20

E (46541) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
E (46541) task_wdt:  - IDLE1 (CPU 1)
E (46541) task_wdt: Tasks currently running:
E (46541) task_wdt: CPU 0: IDLE0
E (46541) task_wdt: CPU 1: stratum admin
E (46541) task_wdt: Print CPU 1 backtrace
```

OsO-1973, 2024-08-23:
Hello
was able to reproduce a crash as discussed with skot
1368, 525, 1166, 2.1.10

open on iPhone 15 pro latest firmware logs, hit show logs and put iPhone on standby
innert 5 mins webgui of Bitaxe is not reachable on LAN anymore

network ping - device is responding
webgui not responding
display on device says see pics
device is not recovering and on display showing evthg alright though ckpool says otherwise

log on ckpool (happened yesterday same at 9am (at 10 p.m i changed some settings) and today at about 8 a.m.

[image]
[image]
[image]
[image]

closed 2024-09-11 by WantClue