bitaxeorg/ESP-Miner #290: Track number of blocks found on dashboard and api

eandersson, 2024-08-14:
This is mostly for fun and wishful thinking, but for my own miners I added a tracker that I stored in nvs_config to track the number of blocks found.

e.g. this is how it shows up in my dashboard
[image]

```
cJSON_AddNumberToObject(root, "blocksFound", GLOBAL_STATE->SYSTEM_MODULE.blocks_found);
```

```
if (diff > network_diff) {
    module->FOUND_BLOCK = true;
    module->blocks_found++;
    nvs_config_set_u64(NVS_CONFIG_BLOCKS_FOUND, module->blocks_found);
    ESP_LOGI(TAG, "FOUND BLOCK!!!!!!!!!!!!!!!!!!!!!! %f > %f", diff, network_diff);
}
```

19201003080114, 2024-08-14:
I vote this is worthy to be included in next update 👍

WantClue, 2024-08-14:
If you suggest implementations open a pull request please.

closed 2024-08-15 by eandersson