bitaxeorg/ESP-Miner #303: Add MQTT Support

ffrediani, 2024-08-20:
MQTT is a pretty versatile, lightweight and simple protocol which allows data export to platforms in order to improve monitoring, specially where there are a reasonable number of devices in order to identify faults or improve efficiency.

MQTT code is well developed for ESP32 and doesn't increase much the binary size or consumes any significant resources in terms of CPU and Memory. Tasmota firmware uses it widely for both ESP8266 and ESP32.

A few new lines in settings would be enough such as: Destination Host, ClientID, Username, Password and Topic.

References: https://tasmota.github.io/docs/MQTT/

skot, 2024-08-20:
This is an interesting idea. We could maybe expand the monitoring/control options while not increasing the HTTP overhead?

Of course we'd need to figure out how to interface with this. Can web apps talk MQTT directly?

Webranger2, 2024-08-29:
That would be a cool idea. This way you could get information in the vis from the iobroker or homeassist.

pixeldoc2000, 2024-09-03:
@Webranger2 
FYI, Home Assistant Integration is already possible via https://github.com/Schnitzel/hass-miner Integration (using pyasic).

MQTT Support +1

closed 2024-10-10 by skot

ffrediani, 2024-10-10:
@skot what happened ? Is it because nobody volunteered so far or you think that is not useful ?

skot, 2024-10-10:
I'm not sure I see the benefit considering we already have a full-featured HTTP API (which we need for AxeOS)

ffrediani, 2024-10-10:
For using HTTP API monitoring systems have to do some minimal work to adapt and consume that information from the device and aggregate it. MQTT is a very simple and well stabilized protocol that is easier to get and aggregate data from multiple devices and get alerted in near real time for events with very little CPU cost and complexity.
Fine. Perhaps someone can catch up this, implement and submit a pull request at some point.

MadNBG, 2024-10-23:
mqtt would be nice. Library for ESP is easy, the dashbords Infos sent in a jason would be enough. Maybe Errors, too..

krecik, 2025-03-21:
If there is an API, same JSON could be publish as a MQTT message, so no need to change this.
MQTT sounds good - easier integration with ex. HomeAssistant