bitaxeorg/ESP-Miner #307: v2.1.10 doesn´t work with Lucky Miner lvl06

TheRockaXXX, 2024-08-25:
Hi,

v2.1.10 doesn´t work with Lucky Miner lvl06. The previous 2.1.8 works without any probs.
On 2.1.10 the mining doesn't start and stays at 0 GH/s.

skot, 2024-08-25:
esp-miner does not support closed source luckyminer hardware.

closed 2024-08-25 by skot

pixeldoc2000, 2024-09-03:
Thas sad, but understandable.

I am in the same boat with a Lucky Miner LVL06 v2.

Mining still works with 2.1.10, but OLED Display, Temperatur and Voltage Sensor does not work anymore.

skot, 2024-09-03:
> That is sad, but somewhat understandable.
> 
> I am in the same boat with a Lucky Miner LVL06 v2. Mining still works with 2.1.10, but OLED Display, Temperatur and Voltage Sensor does not work anymore.

this would be trivial to fix if they had released their hardware changes (as is required by the license).

pixeldoc2000, 2024-09-03:
@skot
True, but it will be complicated to locate the "Vendor" and tell them to do so...

Maybe I find some time to pull the commits apart to figure out what did break things with Lucky Miner.

Is there anything particular I should be looking for?

skot, 2024-09-03:
IMO you should look for a legit Bitaxe

pixeldoc2000, 2024-09-09:
Found out why OLED Display, Temperatur and Voltage Sensor did not work anymore after FW v2.1.8 .

In CVS / NVS config I had a custom devicemodel: `devicemodel,data,string,LV06`.

Everything is fine after I change it to regular Bitaxe 1366 (ultra) settings: `devicemodel,data,string,ultra`.