bitaxeorg/ESP-Miner #310: Phase out support for bitaxeMax in future firmware updates

skot, 2024-08-28:
we can do this with a json file in the releases that specifies the minimum HW revision required to run the update. Then AxeOS needs a change to implement this.

This is paired with the esp-miner firmware migration the bitaxeorg GitHub org

jddebug, 2024-08-29:
Why phase out support? I have a lot of these. They have only recently become stable.

benjamin-wilson, 2024-08-29:
> Why phase out support? I have a lot of these. They have only recently become stable.

The Max was one of the earliest working bitaxes and did not have a large number of sales when compared to the Ultra and Supra. We believe it has reached a state of stability and now is a good time to sunset any more development. Support for older ASICs adds complexity, bloat and maintenance costs that burdens development of new Bitaxe versions.

skot, 2024-08-30:
We may want to phase out new update support for some of the earlier Ultras too.. So this JSON should allow for hardware revision number based filtering

Kubaaaaaaaaaa, 2024-11-05:
> The Max was one of the earliest working bitaxes and did not have a large number of sales when compared to the Ultra and Supra. We believe it has reached a state of stability and now is a good time [...]

But there could be some UI enhancements in the future that even owners of old MAXs could benefits from.