bitaxeorg/ESP-Miner #342: Overheat mode should be more obvious in AxeOS

skot, 2024-09-19:
Right now it is easy to miss that the Bitaxe is in overheat mode in AxeOS. I suggest we put a big red banner at the top of every AxeOS tab.

Barnminer, 2024-09-19:
I agree. When I assist people with issues, they did not even notice this.

skot, 2024-09-20:
I'd also like to have the overheat mode checkbox **checked** by default. Then the user can uncheck it to "Disable overheated mode", causing the whole thing to disappear

skot, 2024-09-26:
@eandersson do you have any time to help with this? I'm hopeless with this front end stuff 😅
the [overheat_mode_improvements](https://github.com/skot/ESP-Miner/tree/overheat_mode_improvements) branch might be a good one to work off

eandersson, 2024-09-26:
> @eandersson do you have any time to help with this? I'm hopeless with this front end stuff 😅 the [overheat_mode_improvements](https://github.com/skot/ESP-Miner/tree/overheat_mode_improvements) branch might be a good one to work off

Haha I am too, but I can take a stab at it this weekend.

PR #359 (2024-09-28): fix: Add overheat alert, knob colors & max, chart dots & label #342
Fix #342 and some fixes to the dashboard knobs and chart
closed 2024-09-28 by skot