bitaxeorg/ESP-Miner #351: Board Version: 0.11 vs. Board Version: 204

unclasp, 2024-09-25:
[image]

With board version 204, only software 2.1.8 works for me. Is that the case with you too? No matter what frequency or current.

19201003080114, 2024-09-25:
I have 204 and v2.1.10 works for me. 
I had v2.1.8 before and v2.1.10 works better.

skot, 2024-09-25:
I've got v2.2.2 on a 102, 201, 202, 204, 401 and 402. So far so good. But I'll keep an eye on it.

sstativa, 2024-09-27:
I don't know if there were any changes on the Pool side or if it was due to the firmware reflash, but 2.1.10 works fine for me now.

First, I tried to install 2.1.10 via the WebUI a month ago, attempting multiple times without success. In the end, I rolled back to 2.1.8.

However, I recently installed 2.2.2, which broke the board (204). I had to reflash the board using a serial console with esp-miner-factory-204-v2.1.10.bin. Not only did I manage to fix the board, but 2.1.10 is also now working for me.

skot, 2024-09-27:
This sounds like the AxeOS based firmware update isn't going so well. Can you tell me more about how you're doing it?

v2.2.2 is working well on my 204

unclasp, 2024-10-02:
I bought the boards on AliExpress. The versions mentioned. They call themselves LuckyMiner 6. I only ever flashed via the web interface. The boards have no connection.

[image]

stealth2600, 2024-10-02:
@unclasp Unfortunately LuckyMiners are knockoff devices. You are likely going to need to contact the manufacturer of these devices for support since they are NOT official Bitaxe devices.

closed 2024-10-10 by skot

unclasp, 2024-12-26:
Solved: I now know exactly what the problem was. @matlen67 here on GitHub seems to have come across the error.

https://github.com/matlen67/LuckyMiner-LV06

_What's this about the board version?
Now, I'm not a huge professional, and I don't feel like going deeper into the code (I just want to do a little mining), but as I see it, after the first board revisions, 0.11/0.12 etc. were still available on the 2xx boards A Mosfet is installed into a voltage regulator which is called Power_Enabled and is controlled via the GPIO10 of the ESP32. There are a few places in the source code where the power management is controlled by the board version. The LuckMiner probably doesn't have the Mosfet because it is based on circuit board revision 0.11 and then won't start because the software is waiting for some kind of feedback. The only strange thing is that after unplugging it 10-20 times it still starts. Maybe a bug? Or intentional? :)_

Try it and fix it!