bitaxeorg/ESP-Miner #395: How to handle CRC5 from asic messages

adammwest, 2024-10-09:
Crc5 is incorrect for returning nonces, all other crc5 are correct (send commands)

according to 
https://github.com/skot/BM1397/blob/master/protocol.md
you need to crc all data in the message, excluding preamble  aa55 and the crc5 bits

in the current implemention, the crc5 bits are included, 
this can easily be corrected

by adding a bits offset to crc5 len parameter inside after the multiplication like so
https://github.com/skot/ESP-Miner/blob/master/components/asic/crc.c#L18
crc5(data,len,offset)
len *= 8
len = len + bit_offset

please note for sending data crc5(data,len,0) is required, this only applies to data from the chip

to get the correct crc5  you need to take the message remove the preamble and replace last byte with 0x80
aa 55 32 b9 bc c3 00 46 9c 
           32 b9 bc c3 00 46 80
buf = [32 b9 bc c3 00 46 80] this is what you create the CRC5 with

to match a crc5
test_val = last byte - 0x80
                         0x9c - 0x80 = 1c

crc5(buf,7,-5)  for bm1397
crc5(buf,9,-5)  for bm136X

I have just checked quickly this works for 
Bm1397 not ESP miner
Bm1368 ESP miner

skot, 2024-10-09:
Ah very nice! I was never able to get the nonce crc5 to work (as you noticed). It does seem a little bit redundant to check the CRC as we would definitely catch any bit errors when we check the nonce difficulty...

But maybe this would save us having to do the much tougher SHA256 calc if there is a bit error..

mutatrum, 2025-03-14:
Fixed by #745

closed 2025-03-14 by mutatrum