bitaxeorg/ESP-Miner #404: add OLED Bitaxe logo screen

skot, 2024-10-10:
When the Bitaxe first turns on we should load the OLED with this sweet logo.

[image]

```
#define IMAGE_HEIGHT 32
#define IMAGE_WIDTH 128

// array size is 512
static const byte bitaxe_splash[]  = {
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x80, 0x00, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x01, 0xc7, 0x00, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x03, 0xef, 0xe5, 0xf0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x07, 0xff, 0xf8, 0xe0, 0x20, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x09, 0xff, 0xf0, 0x40, 0x60, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x10, 0xf1, 0xe0, 0x00, 0xe0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x10, 0x70, 0xe0, 0x83, 0xf8, 0x10, 0x04, 0x08, 0x01, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x20, 0x70, 0xe1, 0xc7, 0xf0, 0x3e, 0x0e, 0x1c, 0x03, 0xe2, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x20, 0x70, 0xe3, 0xf1, 0xe0, 0x7f, 0xcf, 0x9f, 0x07, 0xfc, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x20, 0x70, 0xe3, 0xe1, 0xe0, 0xff, 0xdf, 0x3e, 0x1f, 0xf8, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x21, 0x70, 0xe1, 0xe1, 0xe3, 0xcf, 0x8f, 0x1e, 0x3c, 0xf8, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x12, 0x72, 0xe1, 0xe1, 0xe3, 0xc7, 0x8f, 0x1e, 0x3c, 0x7c, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x0c, 0x77, 0xc9, 0xe1, 0xe0, 0xc7, 0x8f, 0x1e, 0x3c, 0x70, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x77, 0xf9, 0xe1, 0xe0, 0x47, 0x8f, 0x1e, 0x3c, 0x60, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x7f, 0xf1, 0xe1, 0xe0, 0x67, 0x8f, 0x9e, 0x3c, 0xc0, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x71, 0xe1, 0xe1, 0xe0, 0x07, 0x8f, 0xf8, 0x3d, 0x80, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0xf0, 0xe3, 0xf3, 0xf0, 0x1f, 0xcf, 0xfd, 0x7f, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x70, 0xe1, 0xe1, 0xe0, 0x37, 0x97, 0xfe, 0x3c, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x70, 0xe1, 0xe1, 0xe0, 0xe7, 0x83, 0xfe, 0x3c, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x70, 0xe1, 0xe1, 0xe3, 0xc7, 0x8f, 0x3e, 0x3c, 0x04, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x70, 0xe1, 0xe1, 0xe3, 0xc7, 0x8f, 0x1e, 0x3c, 0x08, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x70, 0xe1, 0xe1, 0xe3, 0xc7, 0x8f, 0x1e, 0x3c, 0x18, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x03, 0xf0, 0xe1, 0xe1, 0xe3, 0xc7, 0x8f, 0x1e, 0x3c, 0x78, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x0f, 0xf0, 0xf1, 0xe1, 0xe3, 0xc7, 0x8f, 0x1e, 0x3c, 0x78, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x1f, 0xfc, 0xf1, 0xe1, 0xe3, 0xf7, 0x8f, 0x1e, 0x3f, 0x78, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x39, 0xff, 0x81, 0xf9, 0xfb, 0xff, 0xcf, 0x9f, 0x3f, 0xe0, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x20, 0x7f, 0x01, 0xf1, 0xf3, 0xff, 0x8f, 0x1e, 0x3f, 0xc0, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x20, 0x1e, 0x00, 0xe0, 0xe4, 0xfb, 0x86, 0x0c, 0x4f, 0x80, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x20, 0x04, 0x00, 0x40, 0x40, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x18, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 
  0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
};
```

skot, 2024-10-10:
Shot out to @mutatrum for the Bitaxe bitmap!

mutatrum, 2024-10-17:
A bit more involved than I anticipated, as the pixel alignment is different. I used https://javl.github.io/image2cpp/, with Draw Mode: Vertical, 1 bit per pixel to convert the image.

[image]

Not sure what the variation in brightness in the pixels is all about, could that be burn-in?

skot, 2024-10-17:
I think that looks great! The active area on these displays is frustratingly small and off-center. I have noticed some displays do get this burn-in look. Not sure what's going on there, or if it's permanent.

mutatrum, 2024-11-08:
Fixed by #416

WantClue, 2024-12-01:
Fixed in #416

closed 2024-12-01 by WantClue