bitaxeorg/ESP-Miner #418: Mining not starting above V2.1.8

mountainlove, 2024-10-18:
Up to V2.1.8 there were no problems to connect to "sha256.poolbinance.com". Every FW Version above is not working anymore on this pool. Log:

[image]

skot, 2024-10-18:
Interesting... from your screenshot it looks like it's working. can you provide more log output; ideally starting earlier on?

mountainlove, 2024-10-20:
Sure. Also found something interesting.

With V2.1.8, everything works like expected:

[image]
[image]
[image]
[image]

When i upgrade the Firmware without any Powercycle, i'm able to mining on with V2.3.0 (or any other above 2.1.8):

[image]
[image]
[image]

And after a complete reboot with taking away the Power, the Miner i stucking in that Loop from the first post:

[image]
[image]
[image]

Connection is:
sha256.poolbinance.com / port 8888

Thank you.

sstativa, 2024-10-21:
I had something similar in the past, nothing worked for me except 2.1.8.
Playing around with new firmware I "bricked" my miner and had to flash v2.1.10 using Python via serial port.
This has not only "unbricked" my miner but also fixed issues I had with 2.1.10 in the past. 
Currently, I installed 2.3.0 via Web UI and it works without issues.

mountainlove, 2024-10-21:
I had also to reflash my miner with v2.1.10 first (with wantclue web-flasher). Then i followed similar steps as @sstativa, but i had no Luck to get it mining. For me only a Downgrade to v2.1.8 helps.

matlen67, 2024-10-21:
Um welchen bitaxe geht es denn? Oder ist es zufällig ein LuckyMiner?

LupusPluvia, 2024-11-23:
I have the same problem. It was previously on version 2.3.0. I wanted to update to 2.4.0. It didn't work anymore. I went back to 2.3.0. It doesn't work anymore either. Anything above 2.1.8 doesn't work either. With version 2.1.8 it only takes a few hours and then you have to restart it to get it running again. It's a (Lucky miner lv 6 ) 

The other lucky miner that I haven't touched is still running stable on 2.3.0.

skot, 2024-11-23:
esp-miner does not (and will not) support luckyminer.

LupusPluvia, 2024-11-23:
Thank you! Für die info

sstativa, 2024-11-28:
I run 2.4.0 on LuckyMiner LV06 without issues.

closed 2024-11-28 by skot