bitaxeorg/ESP-Miner #441: Add an AxeOS version somewhere in AxeOS

skot, 2024-11-01:
For one reason or another people are not always getting AxeOS (www.bin) updated when they update esp-miner.bin. Could we put a AxeOS version number somewhere? I suppose this could be the same version number as esp-miner.

STSMiner1, 2024-11-02:
Add a note on the Settings page that both files are required when doing the firmware update.

skot, 2024-11-02:
> Add a note on the Settings page informing the end user that both files are required when doing the firmware update / or when downgrading to an older firmware version before clicking on "Restart".
> 
> 
> 
> Other option is to start naming the files from here on out with their firmware version
> 
> 
> 
> ie 
> 
> 
> 
> esp-miner-fw230.bin
> 
> 
> 
> www-fw230.bin

Yes! We should do all of these things.

mrv777, 2024-11-04:
Maybe make it required that both files are provided before the updating can even start?

alainleroy1963, 2024-11-17:
Hi everyone, maybe im dumb (im not a dev) but is it possible technically to make something like this ?
[image]